autowarefoundation/autoware #1475: waypoint_loader is too complicated

yn-mrse, 2018-08-29:
# New Feature
Simplify and decompose function of waypoint_loader.

yn-mrse, 2018-09-07:
Please give us a frank opinion.

hakuturu583, 2018-10-09:
I think it is a good idea, but I will add new features to my diagnostic library.
It checks the node exists or not  and publish this information.
You can use this information to switch topics.
And, it enables you to switch topics online.

hakuturu583, 2018-10-09:
[image]

hakuturu583, 2018-10-09:
Sorry. I will update like this.

hakuturu583, 2018-10-09:
[image]

closed 2019-02-06 by yn-mrse