autowarefoundation/autoware #1478: Can not make vector map from VectorMapper

DWKE, 2018-08-29:
I made vector map .csv files using VectorMapper tool(https://maptools.tier4.jp/),

but I couldn't see any lane in Rviz.

I followed the guideline of Maptools homepage.

  - Ubuntu version: 16.04
  - ROS version: kinetic
  - Autoware branch:

likui01, 2018-08-31:
hey, how did you get vector.csv and signal.csv files?

kitsukawa, 2018-09-03:
@DWKE Lane is not visualized on Rviz because vector_map_loader does not publish markers for lane.

closed 2018-09-10 by kitsukawa

hamkuchan, 2018-10-01:
@kitsukawa I have another problem, when I try to click "Vector Mapper" button on page https://maptools.tier4.jp/tools/, it shows me the following information:
-------------------------------------------------------------------------------------------------------------------------------
Using the URLconf defined in maptools.urls, Django tried these URL patterns, in this order:

^ ^$ [name='index']
^ ^tools/$ [name='tools']
^ ^login/$ [name='login']
^ ^login_json/$ [name='login_json']
^ ^logout/$ [name='logout']
^ ^contact_us/$ [name='contact_us']
^ ^complete/$ [name='complete']
^ ^prepare/$ [name='prepare']
^ ^error/$ [name='error']
^ ^terms_of_service/$ [name='terms_of_service']
^ ^vector_mapper_description/$ [name='vector_mapper_description']
^static\/(?P<path>.*)$
The current path, tools/104.210.54.222, didn't match any of these.
------------------------------------------------------------------------------------------------------ 
I'm wondering if the vector mapper online tools still available for us?

thanks a lot!

kitsukawa, 2019-01-07:
@Probacal Currently, there are no markers for wayareas. But I think you can see where wayareas covers on RViz using wayarea2costmap node. Is this right? @cirpue49 @amc-nu