autowarefoundation/autoware #1489: ndt_mapping doesn't work fine

hamkuchan, 2018-09-03:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
using ndt_mapping and .bag file can generate a map. 

## Actual Behavior
sometimes became unstable

## Steps to Reproduce the Problem

  1. 
  1.
  1.

## Specifications

  - Ubuntu version:16.04
  - ROS version:Kinetice
  - Autoware branch:Master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

kitsukawa, 2018-09-03:
@hamkuchan Could you describe the situation on more detail? Kind of lidar, places, how to mount the lidar on the vehicle and so on.

hamkuchan, 2018-09-05:
@kitsukawa The lidar we use is ls16 of this company: http://en.leishen-lidar.com/, and we use it to do the mapping of our industrial park, and we mounted the lidar in this way: https://youtu.be/9vvejy-KQs0, and the issue like this: https://youtu.be/8Sm9QFwMsxc.
Thanks a lot!

kitsukawa, 2018-09-10:
@hamkuchan If you have further questions, please feel free to reopen this issue.

ajay1606, 2019-01-10:
@kitsukawa , I am trying to generate map using ndt_mapping as presented in the document file:///home/ajay/Downloads/Autoware_NDT_Mapping_Tututorial.pdf

But something going wrong, i am unable to visualize the points in the rviz. Coudl you please help me to fix this issue ?

Step1: Loading bag file
[image]
Step 2: Run file:///home/ajay/Downloads/Autoware_NDT_Mapping_Tututorial.pdf
[image]
Step3: ndt_mapping
[image]
Step 4
: Run rviz
[image]

Kind Regards,
AJay

kitsukawa, 2019-01-10:
@ajay1606 It is difficult for me to find the solution from the information. Can you share the rosbag with me if possible?

ajay1606, 2019-01-10:
@kitsukawa Thanks for the quick response. https://drive.google.com/open?id=1HsabYoNX5nlZbo266os-cKKSLlyqpsjl
PLease free to use the data found in the above link

Regards,
AJay

kitsukawa, 2019-01-10:
@ajay1606 I looked the rosbag.
You need to run
`rosrun topic_tools relay /velodyne_points /points_raw`
instead of
`rosrun topic_tools relay /front/velodyne_points /points_raw`
because the rosbag contains `/velodyne_points` topic.
I managed to generate the map from the rosbag.
[image]

ajay1606, 2019-01-11:
@kitsukawa Thanks a lot, Greatly appreciate your help.

closed 2019-01-11 by kitsukawa

ajay1606, 2019-01-13:
@kitsukawa Is there any issues with saving the point cloud? I trying to save ndt_map point cloud file, but unable to find the file after it saves.
In the terminal, this message printed "Saved 79620 data points to autoware-190113.pcd."  I checked in the location $.ros/ , But there is no any such file.
If i change the directory file saving, i get message like this "(runtime_manager_dialog.py:191): Gtk-WARNING **: Attempting to set the permissions of `/home/autoware/.local/share/recently-used.xbel', but failed: No such file or directory"

Am i doing anything wrong in saving file ?

kitsukawa, 2019-01-13:
@ajay1606 There is no problem regarding saving the point cloud in my environment. Can you try again to save the file in your home directory and wait for some seconds after clicking `OUTPUT PCD`? You should not uncheck the checkbox for ndt_mapping until the node finish writing data to PCD.

ajay1606, 2019-01-14:
@kitsukawa I have carefully followed these steps, 
1. Run bag file and Pause
2. Enable ndt_mapping in the computing tab (Enable Check)
3. unpause bag file (run for few seconds )
4. Pause the bag file
5. In the ndt_maping, click on app and click on "PCD OUTPUT" 
6. In the terminal, I got the message " Saved 26024 data points to autoware-190114.pcd."
7. Waited for the 10 minutes and uncheck the ndt_mapping.
 
Still, I couldn't able to find the file in the location $home/.ros/

I am wrong in any of the step mentioned above? Could you please suggest me?

Kind Regards,
Ajay

kitsukawa, 2019-01-14:
@ajay1606 The PCD is not output in $HOME/.ros/. If you do not specify the output dirctory, the file will be output in Autoware/ros/ directory by default.

ajay1606, 2019-01-14:
@kitsukawa I I am unable to find the files in Autoware/ros/ and Shome/.ros/ Both of the locations.
Even though I have changed save locations also, I am unable to find the files. 
Also in the PR #533, I understood that the default location is in Home/.ros/, 

Sincerely, looking for your help 
Kind Regards,
Ajay

kitsukawa, 2019-01-14:
@ajay1606 Can you show me the screen capture to see how you operate?

ajay1606, 2019-01-16:
@kitsukawa Have you got any chance to look into the video file I have shared? 
Am I wrong in any of the steps?

kitsukawa, 2019-01-17:
@ajay1606 I checked the video. You are running Autoware on Docker. I am not sure since I have never tried mapping on Docker but saving pcd might not work on Docker. I guess you need to mount a host directory in Docker container. I am sorry if I say something wrong.

ajay1606, 2019-01-17:
@kitsukawa You are right, Yes it is working in with source build (Without Docker) and files are saving in $.ros/.. And I understood the issue with docker as you mentioned. I will try to mount the host directory and update you. 

Thank you so much for your time and valuable comments. 

Regards,
Ajay

ajay1606, 2019-01-21:
@kitsukawa Its working now with docker after hosting a directory. Thank you

HongweiSunny, 2019-03-10:
I worked not in Docker, but I can  not get pcd
 file as the original question. Branch master, KInetic, Ubuntu16.04.
I have follow these stps as [https://github.com/CPFL/Autoware/issues/1489#issuecomment-453881676](url).
Thanks in advance.

elNiNo009, 2020-03-05:
Facing the same problem. Not able to generate NDT map

ajay1606, 2020-03-05:
> I worked not in Docker, but I can not get pcd
> file as the original question. Branch master, KInetic, Ubuntu16.04.
> I have follow these stps as:
> 
> * @kitsukawa I have carefully followed these steps,
> * ```
>   Run bag file and Pause
>   ```
> * ```
>   Enable ndt_mapping in the computing tab (Enable Check)
>   ```
> * ```
>   unpause bag file (run for few seconds )
>   ```
> * ```
>   Pause the bag file
>   ```
> * ```
>   In the ndt_maping, click on app and click on "PCD OUTPUT"
>   ```
> * ```
>   In the terminal, I got the message " Saved 26024 data points to autoware-190114.pcd."
>   ```
> * ```
>   Waited for the 10 minutes and uncheck the ndt_mapping.
>   ```
> * Still, I couldn't able to find the file in the location $home/.ros/
> * I am wrong in any of the step mentioned above? Could you please suggest me?
> * Kind Regards,
> * Ajay
> 
> Thanks in advance.
@HongweiSunny Have you checked the file in default location ($HOME/.ros/)?

ajay1606, 2020-03-05:
> Facing the same problem. Not able to generate NDT map
@elNiNo009 , If you are working with source build version of Autoware, ndt_map out put pcd files will be saved in the default location $HOME/.ros/  PLease check once.

elNiNo009, 2020-03-10:
@ajay1606 The problem is I am not able to visualize.

ajay1606, 2020-03-10:
@elNiNo009 It's very hard to help, without knowing the actual issue. Please share the error message if are getting. Also, check you have chosen **ndt_mapping.rviz**.

elNiNo009, 2020-03-11:
@ajay1606 The problem is same as your problem you commented on jan 10 2019.

ajay1606, 2020-03-11:
@elNiNo009 Regarding the Jan 10 comment, it was because of the Fixed frame issue. In my case, I have used Velodyne lidar, which publishes points in the topic name /velodyne_points, but autoware ndt_map subscribes the points in the topic /points_raw. 
Just by running the following command, /velodyne_points mapped to /points_raw
**rosrun topic_tools relay /velodyne_points /points_raw**
Otherwise, you can change the topic name in the lidar ros package also.

elNiNo009, 2020-03-12:
@ajay1606 
I got the visualisation correct. But I am not getting pcd output. I followed the steps. I have .csv file in $HOME/.ros/ and no file in location I mentioned to save pcd output.

ajay1606, 2020-03-12:
@elNiNo009 if you are working with docker environment, you should mount the directory before saving .pcd file . So Please refer docker page, to understand mounting directory in the docker container.

elNiNo009, 2020-03-12:
@ajay1606 I am working with source build

ajay1606, 2020-03-13:
@elNiNo009 In case of source build version of Autoware, ndt map pcd files will be saved in the location **$HOME/.ros/**. if following steps followed correctly,

1. Run bag file and Pause
2. Enable ndt_mapping in the computing tab (Enable Check)
3. unpause bag file (run for few seconds )
4. Pause the bag file
5. In the ndt_maping, click on app and click on "PCD OUTPUT" 
6. In the terminal, I got the message " Saved 26024 data points to autoware-190114.pcd."
7. Waited for the 10 minutes and uncheck the ndt_mapping.

Please check your procedure again, or upload the video so that i can help you.

munitioner, 2022-09-10:
Hi, did you solve your problem, I have the same problem