autowarefoundation/autoware #1501: [Design] Fusion of Vision Detector, Vision Tracker, and Lidar Detector

kfunaoka, 2018-09-06:
# New Design
[A motif in Design Rules](https://github.com/CPFL/Autoware/wiki/Design-Rules#a-motif) has no design among Fusion of Vision Detector, Vision Tracker, and Lidar Detector because Vision Tracker tracks only the results of Vision Detector.

This issue is where we discuss how the fusion should be.

kfunaoka, 2018-09-06:
Does this connection work?

* vision_detector -> vision_tracker -> range_vision_fusion
* lidar_detector -> lidar_tracker -> range_vision_fusion
* range_vision_fusion -> object_tracker

kfunaoka, 2018-09-07:
I’ve discussed the design with @amc-nu. If any question or suggestion, please let me know.

The conclusion now is:
* vision_tacker is inserted between vision_detector and range_vision_fusion in order to improve the input of range_vision_fusion and (as the rusult) object_tracker.
* object_tracker is not renamed (to e.g., lider_tracker) because it might uses the info (e.g., what the object is) come from vision_detector such as yolo3.
* At this moment, it seems that lidar_tracker between lidar_detector and range_vision_fusion is unnecessary.

[image]

dejanpan, 2018-10-20:
# General Guidelines for Architecture Design
## Definitions
**Deterministic**: Same output assuming the same sensor data input.

## Guidelines
1. The software is organized in modules called nodes which are represented as ROS nodes in code
1. Every node must handle one specific task (separation of concerns).
1. The coupling between nodes should be as low as possible.
1. Nodes are only allowed to communicate with each other over ROS
1. MSG/ files are the interface specification of the nodes. Therefore their design and documentation must be done with great care and foresight.
1. Nodes are not allowed to share threads to reduce coupling on a resource level
1. Nodes must not depend on implementation details of other nodes.
1. Every node has one main thread which runs a while loop.
1. Sequential parallelization inside a node is not allowed. This is done on the ROS level.
1. Parallel parallelization must be done using e.g. TBB or OpenMP
1. Before processing data everything needs to be transformed into the same target frame and target time. Every node must have exactly one triggering topic which triggers a new iteration of the while loop on data received. See [link](https://roscon.ros.org/2018/presentations/ROSCon2018_deterministic_asynchronous_message_driven_task_execution.pdf)
1. Use the timestamp of the triggering data.
1. Deterministically calculate the target time based on the timestamp of the triggering data
1. The access of the system clock is not allowed (std::chrono::system_clock etc). In other words the system clock must not influence the behavior of the system. 
1. Sleeping based on duration or time is not allowed (only waiting for external triggers).
1. rocesses are not used to modularize the software. Nodes are used for this purpose.

## Guidelines C++
1. OOP and RAII paradigms to be used strictly through the code.
1. Exceptions must be used. Return values are not allowed.
1. Non-const reference members are not allowed.
1. The use of points must be technically justified and this justification must be documented.
1. Use references over pointers.
1. Use raw pointers to indicate non-ownership pointers and smart pointers for ownership pointers.
1. Prefer unique pointers over shared pointers.

dejanpan, 2018-10-20:
@kfunaoka @amc-nu couple of comments on this: https://github.com/CPFL/Autoware/issues/1501#issuecomment-419325252:

1. Does the `vision_detector` assume that the image coming in is already [undistorted](http://wiki.ros.org/image_pipeline/CameraInfo)
1. What does `vision_detector` do? I assume it loads a certain CNN network configuration and model? Is it this configurable?
1. Where are `autoware_detection_msgs::ObjectArray` defined? I would like to see what is inside to know what the input and output for `vision_tracker` is
1. I see that this tracker here requires a different input: https://github.com/CPFL/Autoware/tree/develop/ros/src/computing/perception/detection/vision_tracker/packages/vision_beyond_track
1. What does the `range_vision_fusion` do? Fuse bounding boxes from the `lidar detector` and `vision_tracker`? Or does it fuse lidar and image points? In any case this node also needs the TF and we should ask [Brian](https://roscon.ros.org/2018/presentations/ROSCon2018_deterministic_asynchronous_message_driven_task_execution.pdf) how to have non-blocking TF calls
1. For the fusion we should  transform data into the same target frame and target time. This node must have exactly one triggering topic which triggers a new iteration of the while loop on data received also waits on the data from the non-triggering topic
1. I am at this point not sure what is the role of a `fusion_detector`?
1. And why do we have another `object_tracker` at the end?
1. The nodes should be programmed so that they have at least an init and a run state. 
1. We should also use http://wiki.ros.org/bondcpp to be able to monitor the nodes
1. We should use ROS_[INFO|WARN|DEBUG] but then disable it in "production" https://answers.ros.org/question/9627/how-can-i-completely-disable-writing-logs-to-filesystem/?answer=225993#post-id-225993
1. We have a `lidar_tracker` coming up in [Autoware.Auto](https://gitlab.com/AutowareAuto/AutowareAuto/issues/2) 
1. We need to treat map as another sensor, e.g. [see](https://medium.com/@LyftLevel5/https-medium-com-lyftlevel5-rethinking-maps-for-self-driving-a147c24758d6) and also for filtering of objects out or in of a certain region of interest
1. We should start adding long missing features:
    1. object classification (as oppose to only detection)
    1. lane marks recognition (in case we start drifting outside of the lane)
    1. curb and drivable area recognition (drivable are will a polygon and not a bounding box anymore) 
    1. unknown object classification

dejanpan, 2018-10-20:
For the reference I will just add that there are also perception architectures that are built around the grid worlds:

1. http://www.transport-research.info/sites/default/files/project/documents/20130823_164344_22255_paperModellingTrackingDrivingEnvironment_byDanescuothers.pdf
2. http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.707.5047&rep=rep1&type=pdf
3. https://hal.inria.fr/inria-00182004/document
4. https://hal.archives-ouvertes.fr/file/index/docid/295084/filename/08-star-tay-etal.pdf

kfunaoka, 2018-10-21:
I've created a issue #1628 for `General Guidelines for Architecture Design`.

We'll answer Dejan's questions about perception after ready.

kfunaoka, 2018-10-22:
> @kfunaoka what tool did you use for diagram here: #1501 (comment).

@dejanpan I'm using http://en.plantuml.com/. I'd like to know better tool.

I'll look into your comments in this issue and #proj-architecture tomorrow.

kfunaoka, 2018-12-10:
This is the latest version of the component graph for perception module.
https://github.com/CPFL/Autoware/issues/1409#issuecomment-429686913

closed 2020-05-29 by gbiggs