autowarefoundation/autoware #1505: Problems to follow demo with docker

wmpang, 2018-09-07:
I followed most of the steps in demo video. While I found problem when loading the map. Once i try to select the my_map.launch file inside the UI, and click "map" button, there is error message like the following: 

[ERROR] [1536307140.463298956]: Usage:
[ERROR] [1536307140.463368529]: rosrun map_file points_map_loader noupdate [PCD]...
[ERROR] [1536307140.463378018]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} AREALIST [PCD]...
[ERROR] [1536307140.463385394]: rosrun map_file points_map_loader {1x1|3x3|5x5|7x7|9x9} download
[ERROR] [1536307140.685822984]: Usage:
[ERROR] [1536307140.685893618]: rosrun map_file vector_map_loader [CSV]...
[ERROR] [1536307140.685914358]: rosrun map_file vector_map_loader download [X] [Y]
[points_map_loader-3] process has died [pid 1446, exit code 1, cmd /home/autoware/Autoware/ros/devel/lib/map_file/points_map_loader noupdate __name:=points_map_loader __log:=/home/autoware/.ros/log/3f0614dc-b273-11e8-891b-1c4d70603f38/points_map_loader-3.log].
log file: /home/autoware/.ros/log/3f0614dc-b273-11e8-891b-1c4d70603f38/points_map_loader-3*.log
[vector_map_loader-4] process has died [pid 1463, exit code 1, cmd /home/autoware/Autoware/ros/devel/lib/map_file/vector_map_loader __name:=vector_map_loader __log:=/home/autoware/.ros/log/3f0614dc-b273-11e8-891b-1c4d70603f38/vector_map_loader-4.log].
log file: /home/autoware/.ros/log/3f0614dc-b273-11e8-891b-1c4d70603f38/vector_map_loader-4*.log

Can someone give hints to solve this issue? thanks.

enijmax, 2018-09-19:
I had encountered the same issue.
Is there anyone know where goes wrong?

ajay1606, 2018-11-28:
Hello wmpang, 

I am also having a similar issue with custom vector map loading into the Autoware,
I am getting error like this
[vector_map_loader-5] process has died [pid 509, exit code -6, cmd /home/autoware/Autoware/ros/devel/lib/map_file/vector_map_loader /home/autoware/.autoware/data/map/vector_map/lane.csv /home/autoware/.autoware/data/map/vector_map/line.csv /home/autoware/.autoware/data/map/vector_map/node.csv/home/autoware/.autoware/data/map/vector_map/whiteline.csv /home/autoware/.autoware/data/map/vector_map/road_surface_mark.csv /home/autoware/.autoware/data/map/vector_map/area.csv /home/autoware/.autoware/data/map/vector_map/idx.csv /home/autoware/.autoware/data/map/vector_map/crosswalk.csv /home/autoware/.autoware/data/map/vector_map/stopline.csv /home/autoware/.autoware/data/map/vector_map/intersection.csv /home/autoware/.autoware/data/map/vector_map/point.csv /home/autoware/.autoware/data/map/vector_map/wayarea.csv /home/autoware/.autoware/data/map/vector_map/whiteline.csv __name:=vector_map_loader __log:=/home/autoware/.ros/log/f5b3f884-f22e-11e8-ab54-f832e4708d26/vector_map_loader-5.log].
log file: /home/autoware/.ros/log/f5b3f884-f22e-11e8-ab54-f832e4708d26/vector_map_loader-5*.log

Could you please help me to fix this, if you know about this!
Really it will be more helpful for me.

Kind Regards,
Ajay

closed 2020-05-29 by gbiggs