autowarefoundation/autoware #1506: ladybug 5 isoch operation failed

bryanibit, 2018-09-08:
Fill-out only one section depending on whether you are reporting a bug or a new feature.
I changed ladybug driver to another project and abandon ros. When I captured images for a moment, the driver failed and the error message shows ***isoch operation failed***. The problem usually happens after capturing images for 30 mins or longer normally.  But fail finally.

# Bug
## Expected Behavior
Normal capture

## Actual Behavior
Isoch operation fail

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version: 16.04
  - ROS version: 
  - Autoware branch:master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

amc-nu, 2018-09-08:
@bryanibit we use the driver for way longer than 30 minutes. Please make sure you have:
- an appropriate power supply
- a usb 3 card with a recommended chipset (check FLIR’s website)
- a compatible kernel 
- performed the usb setup for isochronous transfers as pointed by FLIR.

closed 2018-09-08 by amc-nu

bryanibit, 2018-09-08:
@amc-nu Could I ask you for specific details? One: what is a compatible kernel? Second: usb 3.0 card is required. Third: usb setup points larging usb_memory to 1000MB.  Thank you so much!

amc-nu, 2018-09-08:
Have a look at the following KBs at the FLIR website:
https://www.ptgrey.com/KB/10685
https://www.ptgrey.com/KB/10359

bryanibit, 2018-09-09:
@amc-nu If I use only ladybug5 camera, do I need to install FlyCapture driver? Or I just need ladybug1.15.deb driver. Thank you!

amc-nu, 2018-09-09:
No need for FlyCapture. A ladybug specific version is already included in the ladybug deb.

bryanibit, 2018-09-09:
Now I use [1 port usb hub](https://www.ptgrey.com/1-port-usb-30-hub-with-screw-locks) to connect 1-meter line and 3-meter line between ladybug5 and PC. I can detect Ladybug5 but initialization error happens with incorrect interface. Do the hub need its driver? 

I connect ladybug5 with 3-meter line directly. The ladybug5 could run ok, but it would be broken after a moment. So I think to add the recommended hub.

amc-nu, 2018-09-09:
We also experienced several problems using hubs. Since then, we connect directly the camera. 
Make sure your external power source provides enough current. We experienced several problems like you mention with smaller power sources.

bryanibit, 2018-09-09:
If I can acquire images, then it prove my mothercard or other firmwares are satisfied with FLIR. However, it fail to acquire images after a moment, that could be power problem. The above is correct?

amc-nu, 2018-09-09:
In your case I think it is worth the shot.