autowarefoundation/autoware #1507: Autoware on Jetson Xavier

kennedywai, 2018-09-11:
Anyone who has plans to run Autoware on Jetson Xavier Developer Kit? We could work together on this.

yk-fujii, 2018-09-11:
Hi @kennedywai 

I have got a Jetson Xavier, and I have a plan to run Autoware on Xavier.
But I will take some time to get started. I am very happy if you can try in advance.

I think to encount the ROS version issue because:
- ROS Kinetic is not present for ubuntu18.04
- Autoware is not support ROS lunar or ROS melodic

kennedywai, 2018-09-11:
@yk-fujii Cool! I'm glad that you are going to try running Autoware on Xavier. Nvidia just released Jetpack 4.0 preview specifically for Xavier and I am not sure that Jetpack 3.3 will support Xavier. Will Tier 4 switch the dev kit from PX2 to Xavier?

kennedywai, 2018-09-18:
@yk-fujii Cool! This is what I have been thinking! The price is just nearly 2x than the TX2 but the specs are way better than TX2.

jwen11, 2018-12-07:
Hi , I recently got a Jetson AGX Xavier and is looking forward to deploy Autoware on it.
Is there any thread I can join to get the latest updates on the porting progress?

kennedywai, 2018-12-08:
@jwen11 Thats good to hear it! Unfortunately I havent been able to afford one since it released. The first thing we should do is run Autoware on Ubuntu 18.04 + ROS Melodic and then Xavier.

IndefiniteBen, 2019-05-07:
Related to #1697

sagar448, 2019-05-09:
Ok any luck? Honestly tried to build ROS Kinetic on Xavier but it failed I believe only melodic works, Autoware I think doesnt support Kinetic, so whats like the next steps?

IndefiniteBen, 2019-05-09:
I haven't actually done this, but my research on how to run Autoware on a TX2 with JetPack 4.2 lead me to the same conclusion that only ROS Melodic will work. Based on what I've found, the issues that were blocking Melodic support are fixed (see my comment on #1697) but it hasn't been moved to release. If you look at [autowarefoundation/autoware/tree/feature/melodic](https://github.com/autowarefoundation/autoware/tree/feature/melodic) you can see the branch with these changes. Pull that branch and follow the build from source instructions? 

tl;dr
Pull [autowarefoundation/autoware/tree/feature/melodic](https://github.com/autowarefoundation/autoware/tree/feature/melodic)
Follow these [instructions](https://github.com/autowarefoundation/autoware/wiki/Source-Build)

gbiggs, 2019-05-09:
The work for melodic still has a couple of issues that are actively being worked on.

https://github.com/autowarefoundation/autoware/pull/2032

kennedywai, 2019-05-11:
Keep up the good work guys! When Melodic is ready, I am willing to be the first to test it!

amc-nu, 2019-05-11:
You can already test Autoware on the Jetson AGX using the melodic branch. #2145

kennedywai, 2019-05-11:
@amc-nu Nice. I am also planning to test simple version of Autoware on Jetson Nano

sagar448, 2019-05-16:
Also just a unrelated question didn't really know where to ask but.....is it only the velodyne lidars that are compatible? Can I just get any of the shelf lidar and autoware would work with that?

gbiggs, 2019-05-16:
@sagar448 Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

mangoSteve, 2019-05-20:
I am trying to build autoware on my Xavier, but it seems that colcon failed when building

`Starting >>> autoware_health_checker                                                
--- stderr: autoware_health_checker                                                 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/executor/__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/task/__init__.py", line 92, in __call__
    return await task_method(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_ros/task/catkin/build.py", line 71, in build
    skip_hook_creation=True, additional_targets=additional_targets)
  File "/usr/lib/python3/dist-packages/colcon_cmake/task/cmake/build.py", line 97, in build
    .format_map(locals())
AttributeError: 'BuildPackageArguments' object has no attribute 'name'
---
Failed   <<< autoware_health_checker	[ Exited with code 1 ]`

I am wondering if some of you have also encountered the error.

Sorry for asking questions here, I tried to follow the support guideline and asked on ROS Answers website, but no one has replied yet. 
https://answers.ros.org/question/323417/autoware-buildpackagearguments-object-has-no-attribute-name/

gbiggs, 2019-05-20:
Please ask your question at ROS Answers.

mangoSteve, 2019-05-20:
> Please ask your question at ROS Answers.

Sorry for asking questions here, I tried to follow the support guideline and asked on ROS Answers website, but no one has replied yet. I am just wondering if someone also encountered the same error here.
https://answers.ros.org/question/323417/autoware-buildpackagearguments-object-has-no-attribute-name/

gbiggs, 2019-05-20:
It sometimes takes a little time to get an answer, and asking in multiple places, especially places you are explicitly asked not to ask at, is more likely to get developers annoyed at you than to get you an answer. Furthermore, you asked your question right before the weekend, so you are even expecting people to give up their personal time to answer your question.

mangoSteve, 2019-05-20:
> It sometimes takes a little time to get an answer, and asking in multiple places, especially places you are explicitly asked not to ask at, is more likely to get developers annoyed at you than to [...]

Apologize for question here, I never expect people to give up their personal time but I admit that I should not be in such a hurry. Sorry about that.

Anyway, thanks for your reply.

closed 2019-05-31 by amc-nu

IndefiniteBen, 2019-06-01:
Closed because it's fixed? Now we can just clone master, follow the source build instructions and it'll work on Ubuntu 18 systems?

gbiggs, 2019-06-03:
It should, but keep in mind that it hasn't been released yet so there may well be bugs.

atyshka, 2019-07-24:
Those who have successfully used autoware on Xavier, have you compiled on device or cross--compiled? Just wondering since on-device compilation can be slow for large projects.

gbiggs, 2019-08-04:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://gitlab.com/autowarefoundation/autoware.ai/autoware/wikis/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

AndreV84, 2020-05-13:
Hi Guys,
I am just wondering if there are installation steps for Nvidia Jetson Xavier AGX.
Thanks

urbansound8K, 2022-04-28:
I could manage to install autoware-ai on xavier nx

AndreV84, 2022-04-28:
@urbansound8K 
Could you share the installation steps, please? To reproduce the setup at Xavier NX?
Thanks

urbansound8K, 2022-04-28:
Source installation. I will share the steps next week. But in general 
I installed jetpack 4.4 then installed its dependencies one by one. I booted the unit from emmc and used SSD for installation