autowarefoundation/autoware #1508: can not launch runtime manager

hamkuchan, 2018-09-11:
I could launch runtime manager before, but yesterday it suddenly didn't work, and then I modified the "run" file in ~/Autoware/ros, I modified the last line of "run" with the following code:

${TERMINAL} ${RUNMGR_DISPLAY_OPTION} ${OPTION_WORKING_DIR}=${MY_PATH} ${OPTION_COMMAND}="bash -c 'source ./devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py; sleep 1000;'"

and the I ran ./run in terminal, and got the following message:
/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2889: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()
loading param.yaml
loading qs.yaml
Traceback (most recent call last):
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 3443, in <module>
    app = MyApp(0)
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8628, in __init__
    self._BootstrapApp()
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk2/wx/_core.py", line 8196, in _BootstrapApp
    return _core_.PyApp__BootstrapApp(*args, **kwargs)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 2890, in OnInit
    frame_1 = MyFrame(None, wx.ID_ANY, "")
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 147, in __init__
    self.setup_buttons(self.qs_dic.get('buttons', {}), self.qs_cmd)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 725, in setup_buttons
    pdic = self.load_dic_pdic_setup(k, d2)
  File "/home/nvidia/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 1979, in load_dic_pdic_setup
    pdic = self.load_dic.get(name, {})
AttributeError: 'NoneType' object has no attribute 'get'

Could any one help to have a look at it?
thanks 

# Bug
## Expected Behavior
Could launch runtime manager.

## Actual Behavior
Could not launch runtime manager.

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:16.04
  - ROS version:Kinetic
  - Autoware branch:Master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

hakuturu583, 2018-12-06:
I think you are modified and write something worng in runtime_manager_dialog.py or include or read file from runtime_manager_dialog.py.

GODGANG4885, 2019-01-21:
I have such a problem, I reinstalled catkin and solved it. Please reinstall catkin and try again
$ cd ~/Autoware/ros/src
$ catkin_init_workspace
$ cd ../
$ rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
$ ./catkin_make_release

kfunaoka, 2019-01-28:
param.yaml in runtime_manager might be broken

sumenpuyuan, 2019-11-21:
I have encountered the same problem. Is the problem solved?

closed 2020-05-29 by gbiggs

CCTJB2, 2020-09-15:
Such Attribute error for "NoneType" objects most likely is due to the fact that the scripts are unable to reference the correct target objects or files since the path(s) is missing or incorrect.

In my case, param.yaml file under the directory of autoware runtime manager was completely empty. I was successful at solving the above problem by deleting the param.yaml file and running the autoware runtime manager again, it will generate the param.yaml by itself and able to open up the runtime manager normally again. 

I was suggested by my colleague that recompilation would also work to replace all the default files (including the param.yaml in this case) with fresh install files, without affecting other custom created files. I did not try this myself though.

Remember to back up your files (create a whole zip folder maybe) before you do the above, especially if your param.yaml file is not empty like mine (I had nothing to loose).