autowarefoundation/autoware #1509: Can't Run Autoware Docker on master branch

harderthan, 2018-09-12:
# Bug
## Expected Behavior
Build Autoware Docker and Run it.

## Actual Behavior

> Error response from daemon: pull access denied for autoware-kinetic, repository does not exist or may require 'docker login'

```console
HOME/Autoware/docker/generic$ sudo sh ./run.sh kinetic
Use kinetic
Shared directory: /home/root/shared_dir
Using default tag: latest
nvidia-docker | 2018/09/12 14:44:16 Error: Error response from daemon: pull access denied for autoware-kinetic, repository does not exist or may require 'docker login'
```

## Steps to Reproduce the Problem

  1. Follow this [guide](https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker#step2-autoware-docker-run)

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: master

harderthan, 2018-09-12:
After changing branch from master to 1.7.0, It is working good.

harderthan, 2018-09-13:
I fixed this issue.

The problem is that I didn't download Autoware Docker image from Docker hub.
I reference this [page](https://ai4sig.org/2018/07/docker-for-autoware/)

`$ docker pull autoware/autoware:1.7.0-kinetic`

By the way, I was not able to find this step on [Autoware installation guide](https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker).
Did I wrong way to use autoware?

closed 2018-09-13 by harderthan

kargarisaac, 2019-03-29:
Did you find a way?

harderthan, 2019-03-29:
@kargarisaac 
I written the solution on upper description, check that.