autowarefoundation/autoware #1521: The pillars of CPU's and memoryt don't show anything

AthpapAutowareUser, 2018-09-16:
Every time I launch Autoware, the Runtime Manager runs but still the pillars of cpus and memory don't show anything. It looks that the python script missed something, Below is the printing in the new terminal that pops-up:
 `
/home/demo/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2889: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()
loading param.yaml
loading qs.yaml
loading setup.yaml
['rosparam', 'set', 'tf_x', '1.2']
['rosparam', 'set', 'tf_y', '0.0']
['rosparam', 'set', 'tf_z', '2.0']
['rosparam', 'set', 'tf_yaw', '0.0']
['rosparam', 'set', 'tf_pitch', '0.0']
['rosparam', 'set', 'tf_roll', '0.0']
['rosparam', 'set', 'localizer', 'velodyne']
loading map.yaml
loading sensing.yaml
loading computing.yaml
loading interface.yaml
loading data.yaml
loading simulation.yaml
['rosparam', 'set', '/use_sim_time', 'false']
loading status.yaml
loading state.yaml
loading topics.yaml
top: bad delay interval '0.1'
Exception in thread Thread-8:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/demo/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 1422, in top_cmd_th
    s = subprocess.check_output(['sh', '-c', 'env COLUMNS=512 top -b -n 2 -d 0.1']).strip()
  File "/usr/lib/python2.7/subprocess.py", line 574, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['sh', '-c', 'env COLUMNS=512 top -b -n 2 -d 0.1']' returned non-zero exit status 1
`

AthpapAutowareUser, 2018-09-21:
Yes, My Ubuntu OS is in English from scratch.From the very beginning.

udeto, 2019-07-07:
Did you solve this problem? I am facing the same issues.

closed 2019-07-07 by gbiggs