autowarefoundation/autoware #1524: calibration publisher gives error, cmd_param required

shoaibazam, 2018-09-19:
[image]
calibration publisher gives the shown error, when the calibration file path is given.

simon750324, 2018-10-02:
Are you sure your file path is right?
Maybe your file path includes special characters.

shoaibazam, 2018-10-03:
@simon750324 Thanks for your reply. The file path is correct and it contains no special character.

simon750324, 2018-10-03:
BTW, is there no "space" or any character which is not English?

simon750324, 2018-10-03:
I find your image topic is "camera/usb_cam1/image_raw".
I think you may have 2 or more cameras.
If you have 2 or more cameras, you need give them calibration file for each.
If you only give camera1 the calibration file but no for camera2, you will get this error.

simon750324, 2018-10-03:
You need to change "Camera ID" and assign calibration file to other camera.
[image]

shoaibazam, 2018-10-07:
@simon750324 I am still getting the errors. I used the one camera and give the calibration file and it still gives me the same error.

simon750324, 2018-10-08:
Can you provide the total image which you configure the setting in calibration_publisher?

closed 2020-05-29 by gbiggs