autowarefoundation/autoware #1529: ndt_matching with GPS using custom map

javcursor, 2018-09-20:
Hi, I've done the suggested changes #492 and #1127, the conversion seems to have coherent values of X,Y,Z after check the sign of latitude and longitude. I've added a static transform between base_link and gps because they was separated.
I want know why the baselink pose is orbiting randomly over the map.
I let a video with the actual behavior, the dataset and my src folder with modifications (https://www.dropbox.com/sh/2mv422ww4y2rh1b/AABagU-tDogSLqoATnGbU7t4a?dl=0):
I have had modify plane 19 with my world coordinates (lat,lon) and the origin for geographicLib operations is the first coordinate of my map.
- geo_pos_conv.cpp
- nmea2tfpose_core.cpp
- ndt_matching.launch

Any help is welcome.

## Specifications
- Ubuntu 16.04
- ROS Kinetic 
- Autoware 1.8

en-sevilla, 2018-11-09:
Hi @javcursor , did you find a solution? we are trying to use Autoware in Europe but it is really hard to make it works with the issue that you are asking for. 

Any help? someone had been able to transform between UTM-ECEF and making autoware works in Europe? 

Thanks

wangmygithub, 2019-09-18:
I had the same problem ,and I made my gps work by making the fitness_score equal to 300,rather than 500 in ndt_matching.cpp,   gnss_callback function.  You can think of the fitness_score as the maximum error value of base_link and gps. Within the scope of  the fitness_score, It is normal for base_link and gps frame to be separated.

closed 2020-05-29 by gbiggs