autowarefoundation/autoware #1537: Unable to find image 'autoware/autoware:1.8.0-kinetic' locally docker: Error response from daemon: manifest for autoware/autoware:1.8.0-kinetic not found.

lonfr, 2018-09-25:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS version:
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

jacks808, 2018-09-25:
currently Autoware is not support 1.8 in docker. you can use  `1.7.0-kinetic` to start autoware

lonfr, 2018-09-25:
According to your suggestion,this problem is solved,Thank you!

closed 2018-10-02 by kfunaoka

Srivatsa-M-Hegde, 2018-10-18:
I am facing the same issue. Followed the Docker installation instructions as in: https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker (Using the Autoware Docket setup Cast 1)

For the final instruction
`$ sudo sh run.sh kinetic`

I am getting the following error:
`Use kinetic
Shared directory: /home/root/shared_dir
Unable to find image 'autoware/autoware:1.8.0-kinetic' locally
docker: Error response from daemon: manifest for autoware/autoware:1.8.0-kinetic not found.
See 'docker run --help'.`

From going through similar issues here, I see that I have to use 1.7 instead of 1.8. How do I do this downgrade? Just an edit in the command (and run.sh file) or should I start a fresh installation with an older build?

Thanks

yuqli, 2018-11-06:
Followed @jacks808 's comment I modified the last line in run.sh and it seems to work fine. I mean just change the line ` autoware/autoware:1.8.0-kinetic` to `autoware/autoware:1.7.0-kinetic` and it works.

Here is the link to the original documentation : https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker

On another note, since the documentation has an error, maybe you guys can make it clear in the quick start guide as well?

kfunaoka, 2018-11-06:
Sorry, I updated the documentation.

yuqli, 2018-11-06:
Wow thanks so much for the response. I wasn’t expecting anyone to really notice... thanks! 

Sent from my mobile device. Please excuse brevity and typos, thanks!