autowarefoundation/autoware #1538: Invoking "make -j4 -l4" failed

kafe6, 2018-09-25:
[  2%] Linking CXX executable /home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/example_opencv/find_features
CMakeFilesCMakeFiles/find_red_pixels.dir/src/find_red_pixels.cpp.o�/find_features.dir/src/find_features.cpp.o：在函数‘ImageConverter::ImageConverter(ros::NodeHandle&)’中：
find_features.cpp:(.text._ZN14ImageConverterC2ERN3ros10NodeHandleE[_ZN14ImageConverterC5ERN3ros10NodeHandleE]+0x3f8)：对‘cv::namedWindow(cv::String const&, int)’未定义的引用
CMakeFiles/find_features.dir/src/find_features.cpp.o：在函数‘ImageConverter::~ImageConverter()’中：
find_features.cpp:(.text._ZN14ImageConverterD2Ev[_ZN14ImageConverterD5Ev]+0x35)：对‘cv::destroyWindow(cv::String const&)’未定义的引用
CMakeFiles/find_features.dir/src/find_features.cpp.o：在函数‘ImageConverter::imageCb(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)’中：
find_features.cpp:(.text._ZN14ImageConverter7imageCbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE[_ZN14ImageConverter7imageCbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE]+0x800)：对‘cv::imshow(cv::String const&, cv::_InputArray const&)’未定义的引用
find_features.cpp:(.text._ZN14ImageConverter7imageCbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE[_ZN14ImageConverter7imageCbERKN5boost10shared_ptrIKN11sensor_msgs6Image_ISaIvEEEEE]+0x828)：对‘cv::waitKey(int)’未定义的引用
��collect2: error: ld returned 1 exit status
在函数‘ImageConverter::imageCb(boost::shared_ptr<sensor_msgs::Image_<std::allocator<void> > const> const&)’中：
find_red_pixels.cpp:(.text+0x6cb)：对‘cv::imshow(cv::String const&, cv::_InputArray const&)’未定义的引用
find_red_pixels.cpp:(.text+0x6f3)：对‘cv::waitKey(int)’未定义的引用
CMakeFiles/find_red_pixels.dir/src/find_red_pixels.cpp.o：在函数‘ImageConverter::ImageConverter(ros::NodeHandle&)’中：
find_red_pixels.cpp:(.text._ZN14ImageConverterC2ERN3ros10NodeHandleE[_ZN14ImageConverterC5ERN3ros10NodeHandleE]+0x3f8)：对‘cv::namedWindow(cv::String const&, int)’未定义的引用
CMakeFiles/find_red_pixels.dir/src/find_red_pixels.cpp.o：在函数‘ImageConverter::~ImageConverter()’中：
find_red_pixels.cpp:(.text._ZN14ImageConverterD2Ev[_ZN14ImageConverterD5Ev]+0x35)：对‘cv::destroyWindow(cv::String const&)’未定义的引用
learning_ros/Part_3/example_opencv/CMakeFiles/find_features.dir/build.make:128: recipe for target '/home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/example_opencv/find_features' failed
make[2]: *** [/home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/example_opencv/find_features] Error 1
CMakeFiles/Makefile2:19928: recipe for target 'learning_ros/Part_3/example_opencv/CMakeFiles/find_features.dir/all' failed
make[1]: *** [learning_ros/Part_3/example_opencv/CMakeFiles/find_features.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
collect2: error: ld returned 1 exit status
learning_ros/Part_3/example_opencv/CMakeFiles/find_red_pixels.dir/build.make:128: recipe for target '/home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/example_opencv/find_red_pixels' failed
make[2]: *** [/home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/example_opencv/find_red_pixels] Error 1
CMakeFiles/Makefile2:19867: recipe for target 'learning_ros/Part_3/example_opencv/CMakeFiles/find_red_pixels.dir/all' failed
make[1]: *** [learning_ros/Part_3/example_opencv/CMakeFiles/find_red_pixels.dir/all] Error 2
[  2%] Linking CXX executable /home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/irb120_ctl_interface/test_traj_sender_irb120
[  3%] Linking CXX executable /home/zhangkaifu/ros/kinetic/ros_ws/devel/lib/irb120_ctl_interface/irb120_ctl_interface
[  3%] Built target test_traj_sender_irb120
[  3%] Built target irb120_ctl_interface
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
###################################################################################

 Ubuntu version:16.04.3

I run the command   catkin_make
and during the compilation I get that error.

Can any one please help me ?

yk-fujii, 2018-09-26:
This log looks like a build of ROS, not Autoware.
Are you compiling Autoware?

simon750324, 2018-10-08:
You referenced the opencv libraries but it didn't link to the right opencv libraries. I think you need to check the opencv version which you used first and then check the libraries path is right.

amc-nu, 2018-10-08:
@yk-fujii is right. Compilation is failing on code unrelated to Autoware.
I can imagine you just have some other packages inside the same workspace. Those are causing the problems.
I’m closing this, since the error is not caused by Autoware’s code.

closed 2018-10-08 by amc-nu