autowarefoundation/autoware #1541: Add vehicle interface for Autoware

Mani-Sadhasivam, 2018-09-27:
# Vehicle Interface for Autoware

This issue is used to track the progress of Vehicle Interface design for Autoware

ubercool2, 2018-10-09:
@Mani-Sadhasivam can you maybe move the work you have been doing with @jpsamper2009
 in slack to here so it gets more visibility?

Mani-Sadhasivam, 2018-10-10:
@ubercool2 That's the idea behind creating this issue. Will post the vehicle interface API proposed by @jpsamper2009 after we are done with the internal review.

Mani-Sadhasivam, 2018-10-15:
Here is the initial proposal of the Vehicle Interface API. This API is authored by @jpsamper2009 and reviewed by the Linaro-Autoware team.

The objective of this vehicle interface API is to make the process of adding vehicle support a lot easier for the OEM's. Hence the core part (Vehicle Abstraction) is created as a Library which in turn will be utilized by the vehicles.

Feedback/Review is always welcomed :-)

https://autoware-developer.slack.com/files/U71FUS6BU/FD70AK685/vehicleabstraction_api.pdf

kfunaoka, 2018-10-16:
This is a great improvement of vehicle interface because there is nothing until now!

Would you answer my question? Because I'm not so familiar with vehicle interface,
it might not make sense.

Should we keep public accesses to raw msg (e.g., raw_msg/setRawAxCommand/setRawSteerAngle/etc..) in concrete CanMsg classes such as CommandsMsg? If there is no use case where the user code of concrete CanMsg classes uses raw byte array, it might be possible to make raw accesses private.

Mani-Sadhasivam, 2018-10-16:
@kfunaoka Good question! We haven't sorted out the public and private members yet and it makes sense to do it now. Will update the API with private and public members.

dejanpan, 2018-10-17:
@Mani-Sadhasivam great work. I took the liberty to exchange the link to the doc in slack with an actual pdf uploaded to this issue since download from slack did not work for me.

Since I am in the same group as @jpsamper2009 the design makes sense to me. 

@jpsamper2009 can you double check it maybe?

@JWhitleyAStuff would you mind reviewing the design? Thx

jpsamper2009, 2018-10-18:
@JWhitleyAStuff 

> 1. I assume this is going to be written in C++ since C++14 is the ROS2 standard. Many of the class member variables and functions are defined as more C-like (raw pointers instead of smart pointers, structs instead of classes, etc.). Was this intentional?

Not intentional, it was just easier to say `uint8_t * foo_ptr` than `SmartPointer<uint8_t> foo_ptr` in the examples. The proposal and anything I say is meant to be descriptive and not prescriptive.

> 2. In the concrete implementation examples of your CanMsg class, you appear to store both a raw and a physical version of the data. This takes essentially double (or more, depending on the types used for the particular CAN Message's implementation) the storage for storing the same data in different formats. Since you will have to implement a conversion function for storing/retrieving the data anyway, why not store just the physical data and use formulas to convert back and forth from raw (or vice versa)? In other DBC-based implementations I've even seen CanSignal-type class which contains properties such as:
Factor
Offset
toPhys()
toRaw()
fromPhys()
fromRaw()
Name
StartBit
ByteOrder
This way the CanMsg can simply contain an array (or Vector) of CanSignals and the implementation isn't quite as arduous. Plus, the buildRawBytes() or similar function for creating the actual CAN payload can be automated at the abstract level instead of having to implement a custom function to do this for every CAN message. Just some thoughts.

1. The idea was to make the `get` functions fast, offloading all the computation to the `set` functions
    - The logic is that `get` is called when the information is needed, while `set` gets called periodically as messages are read from CAN or as the node prepares the message to be sent.
2. By having a continous `raw_msg`, it is possible to directly copy the whole block of memory (8 bytes for a regular CAN message) to the data field of the CAN message, making the `getRawMsg()` function fast.
    - If we were to store only one of the two (raw or physical values), I would vote for the raw message
    - Further abstracting the signals into their own classes will hinder using a simple memcpy to generate the CAN data field

> 3. Your phys_msg struct implies that it can only store doubles and floats. This wastes a lot of memory when you're storing (even) 50 CAN messages with loads of single-bit flags. Since Bool is usually implemented as a single byte under the hood and a double would be storing 8 of them (on x86), it seems to make sense to at least have double, float, and bool. Most other types could be stored as floats to save memory but your computational cost will be slightly higher too because of the floating-point math. Depends on how memory/computation-conscious you want to be with the implementation. A good example implementation can be found at https://bitbucket.org/tobylorenz/vector_dbc/src/master/.

You only need the phys_msg struct for values that actually have float values. Bools are already stored in the raw_msg struct, so there is no need to duplicate the information in the phys_msg struct. As I see it, there are three types of signals: bools (single bits), integers (no scaling required), and physical values. `bools` would have `set` and `is_set` functions, `integers` would have `set` and `get` functions, and physical values would have `setRaw`, `getRaw`, `setPhys`, `getPhys` functions.

> 4. You may want to consider implementing the CAN interface read function as a callback listener instead of using a polling mechanism. I know both Kvaser and SocketCAN support using callbacks on receive (for an example, see the Listener class in ros-industrial's SocketCAN implementation here: https://github.com/ros-industrial/ros_canopen/blob/melodic-devel/socketcan_interface/include/socketcan_interface/interface.h). This helps reduce any latency induced by the polling period, though it does mean that your application would necessarily become multi-threaded and somewhat non-deterministic.

If we want to multi-thread, maybe wait_sets are better since they are more similar to what CAN is doing: messages are expected at a certain cycle time; after a certain number of periods (e.g 5 cycles) with no messages, there is an error.

> 5. Lastly, you may simply want to consider sending the CAN messages as can_msg/Frame instead of having the raw messages read by an implementation that essentially becomes part of the Autoware stack. 

I'm not sure I completely understand this comment, but what I think you are trying to say is that we could have a ROS2 node that is just receiving raw CAN messages and forwarding them as ROS2 messages with another node processing the actual information. 

I would actually argue that there is no value in having the raw CAN messages in ROS2 message format: 1. how often do you really need to be looking at raw CAN information? And if you do, why aren't you just tapping directly into the CAN bus using a much more powerful software tool from Kvaser or Vector?
2. There is added overhead both for the CAN-related nodes and for other nodes using the DDS bus when you are publishing the raw messages on the DDS bus

> This helps separate the requirements (e.g. don't require the linuxcan SDK if you're using SocketCAN) of the low-level APIs from the Autoware stack.

The idea of the CanDriver class is precisely to abstract away these requirements. Only if your vehicle abstraction needs to use a KVaser device, will you have a dependency on the linuxcan SDK.

JWhitleyAStuff, 2018-10-18:
Completely understand your reasoning for 1 and 2.

> You only need the phys_msg struct for values that actually have float values.
I would disagree with this statement. In many cases, there is scaling and/or offset required but the result is still an integer. In these cases would you still store a separate physical value for every int or just do the conversion in both directions? e.g. a temperature value.

Understood on 4 as well.

The major reason that having raw CAN data is valuable: rosbags. At the very least, having the option to output the raw CAN data to a topic would be recommended. When you are troubleshooting a stack as big and complex as this, having visibility into every layer is a near necessity.

I don't understand your reasoning on the second-to-last paragraph here. Are you making the assumption that users will choose either Kvaser or SocketCAN (or any other CAN library) at build-time? Because I think this is a flawed assumption. Within our development team, we use a mix of both and the tools available for both have advantages and disadvantages. If you assume that a user needs to make this decision at build-time, you assume that it will never change over the lifetime of the user's project. However, abstracting the raw CAN reader/writer into it's own node provides both visibility and flexibility. As for the performance implications, since ROS2 allows you to make the decision of having nodes use shared memory to pass data back and forth (and this can even be done at run-time), the performance difference is negligible - though I do think this feature is not 100% complete as of this writing.

jpsamper2009, 2018-10-18:
@JWhitleyAStuff 

> I would disagree with this statement. In many cases, there is scaling and/or offset required but the result is still an integer. In these cases would you still store a separate physical value for every int or just do the conversion in both directions? e.g. a temperature value.

Sure, you could have integer values in the phys_msg too---notice, however, that if there is scaling, you are likely going to be seeing weird behavior from integer division.

What I was really trying to say is that every raw signal didn't need a corresponding physical signal (which is what was implied by "This wastes a lot of memory when you're storing (even) 50 CAN messages with loads of single-bit flags.")

> The major reason that having raw CAN data is valuable: rosbags. At the very least, having the option to output the raw CAN data to a topic would be recommended. When you are troubleshooting a stack as big and complex as this, having visibility into every layer is a near necessity.

Again, trying to analyze CAN data at the ROS level seems futile considering that KVaser and Vector have a lot more powerful tools to record and analyze CAN data. I have seen a lot of people record CAN data at the pub-sub level, but have never seen somebody actually use this data...

Recording CAN data feels like recording raw UDP packets in ROS2 format: it could be useful when debugging a driver, but for everything else, you probably want the parsed data (e.g. the PointCloud message or a VehicleState message). Even for debugging a driver, you would need to play the data in DDS where a node receives the message, puts it into CAN (or UDP) format such that the driver can read the data from the CAN (or UDP) bus and parse it:

```
Record:
Vehicle ---CAN---> CanDriver ---DDS----> rosbag
Replay:
rosbag ---DDS---> CanDriver ---CAN----> CanDriver ---> DDS

instead of:

Record:
Vehicle ---CAN---> Kvaser/Vector recorder
Replay:
Kvaser/Vector re-player ---CAN---> CanDriver
````

The ideal solution would be to have rosbag be able to syncronize with other data formats (pcaps, etc.) and their recorders/re-players, but sadly rosbag does not have this feature as far as I know. 

I like to treat the vehicle as any other sensor/actuator for which I need to write a driver: the driver is responsible for taking care of all the nitty-gritty steps of opening ports, checking (and maintaining) the connection, parsing raw data, and packaging the data in a manageable way for other nodes. Once the driver is functional and fast enough, I never need to look at the nitty-gritty again.

All that being said, the vehicle abstraction could go as far as publishing the raw CAN data, but again, I'm not convinced anyone will actually use the CAN data in this format.

> I don't understand your reasoning on the second-to-last paragraph here. Are you making the assumption that users will choose either Kvaser or SocketCAN (or any other CAN library) at build-time?

My assumption was that for a given vehicle, you aren't switching between CAN devices. If switching devices is a use case, we could have a CanDriverFactory which loads the correct CANDriver based  on a parameter set when the VehicleAbstraction is instantiated. The vehicle abstraction will already need to take accept a set of parameters at initialization time, so there is no reason the CAN device couldn't be one of these. **Side note:** there will also need to be a VehicleAbstractionFactory so that the same software can be installed and run in two different cars (loading a different vehicle abstraction in each, naturally).

Mani-Sadhasivam, 2018-11-02:
@jpsamper2009 What is the latest here? Are you working on class definitions as requested by @kfunaoka?

dejanpan, 2018-11-04:
@Mani-Sadhasivam where did @kfunaoka request this? 

As far as I remember we agreed that you will work on the implementation after we agreed on the design. And since there was no more reply from @JWhitleyAStuff I assume that above proposal by @jpsamper2009 stands. 

LMK if you think differently.

kfunaoka, 2018-12-04:
I agree the direction led by @Mani-Sadhasivam and @jpsamper2009 :)

I just asked the API usage since to make API public is a big decision at creating API, generally speaking.

kfunaoka, 2018-12-19:
@Mani-Sadhasivam socketcan_interface and ros_canopen's msgs are useful to this interface?
http://wiki.ros.org/socketcan_interface
https://github.com/ros-industrial/ros_canopen/tree/melodic-devel/can_msgs

JWhitleyAStuff, 2018-12-19:
It's worth noting that a handful of DBC->C++ Class Generators exist that would automate portions of creating the API for a new vehicle. Here is one example: https://bitbucket.org/tobylorenz/vector_dbc/src/master/

closed 2020-05-29 by gbiggs