autowarefoundation/autoware #1574: Pixel-Cloud fusion node coordinate frame problem

wmpang, 2018-10-09:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
plot fused points with color

## Actual Behavior
nothing appears

## Steps to Reproduce the Problem

  1. publish camera calibration
  2. open lidar and camera
  3. launch pixel cloud fusion node
  4. visualize in Rviz

## Suspected problem
It is found that x and y are regarding as the horizontal and vertical directions in side the code. However, in ROS, z is supposed to be upward direction. (correct me if I am wrong)

amc-nu, 2018-10-09:
I believe you might face problem in your frame_id setting, the coordinate frame system is rotated and transformed using the registered TF in the tree.
We set the Z axis to be ointing upwards in the Lidar frame. As for the camera, it depends on your calibration setting. Check if there are messages in the terminal for TF errors.

If you require extra help, share a rosbag with either the TF included or attach your extrinsic calibration file.

wmpang, 2018-10-09:
As I am new to ROS and Autoware programming, may I know more about the frame_ids that are involved in this pixel cloud fusion node? As I had tried hard to trace where to set or publish those frame_ids.  Also, I will try providing my rosbag tomorrow.

amc-nu, 2018-10-09:
The frame ids depend on your setup. The fusion is performed using the frame ids contained in the camera and  lidar messages. These are not fixed.

amc-nu, 2018-10-18:
Please have a look at this, regarding on the coordinate frames in OpenCV.
https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

closed 2018-10-18 by amc-nu

shoaibazam, 2018-10-21:
@amc-nu hi
i am following the same steps as mentioned above. 
getting the following errors.
"yuyv" passed to lookupTransform argument target_frame does not exist.
 [pixel_cloud_fusion] Waiting for Camera-Lidar TF and Intrinsics to be available.