autowarefoundation/autoware #1575: Image distortion effect when calling the camera by "roslaunch autoware_driveworks_interface gmsl_interface.launch"

piantou, 2018-10-09:
After execute "roslaunch autoware_driveworks_interface gmsl_interface.launch" in the PX2, and execute "rosrun rviz rviz" to show the image, which is about the "/image_raw" topic ,the image is distorted, the red color of the image cannot be displayed and it is displayed in yellow,which can be see below. 
The first picture is the image displayed by rviz, and the second picture is the picture taken by the mobile phone.

[image]

[image]

piantou, 2018-10-09:
@Abraham Monrroy

wmpang, 2018-10-09:
Yes. I found the same issue, while when I launch another ROS package (https://github.com/DavidGogh/gmsl_driver_multiple) to access the gmsl camera,ithe color are correct.

piantou, 2018-10-09:
And how to resolve it?@wmpang

wmpang, 2018-10-10:
I haven't tried to figure out the problem inside autoware_driveworks_interface. I simply compile and launch "https://github.com/DavidGogh/gmsl_driver_multiple" instead.

amc-nu, 2018-10-10:
@piantou this is caused due to the camera sensor you are using. It is not compatible with the current node. you might use other drivers.

closed 2018-10-10 by amc-nu

piantou, 2018-10-10:
> @piantou this is caused due to the camera sensor you are using. It is not compatible with the current node. you might use other drivers.

like using "https://github.com/DavidGogh/gmsl_driver_multiple"??? There are other way to resolve it ?

amc-nu, 2018-10-10:
yes, that is one option. You could clone the repo inside Autoware sensing/driver packages and recompile. to use everything in a single workspace.