autowarefoundation/autoware #1577: Invoking "make -j8 -l8" failed

4n00815, 2018-10-10:
Hello everyone!

I encountered an error while trying to compile (./catkin_make_release). Compiling reaches around 73% and just stop.

[ 73%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/dt_GPU.cpp.o
[ 73%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/multithreading.cpp.o
[ 73%] Building CXX object computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/get_boxes.cpp.o
/home/strax90/Autoware/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/gpu/get_boxes.cpp: In function ‘FLOAT* dpm_ttic_gpu_get_boxes(FLOAT**, FLOAT*, int*, GPUModel*, int*, FLOAT*, FLOAT)’:
/home/strax90/Autoware/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/gpu/get_boxes.cpp:1226:34: error: call of overloaded ‘abs(FLOAT&)’ is ambiguous
  FLOAT abs_threshold = abs(thresh);
                                  ^
In file included from /usr/include/c++/6/cstdlib:75:0,
                 from /usr/include/c++/6/bits/stl_algo.h:59,
                 from /usr/include/c++/6/algorithm:62,
                 from /home/strax90/Autoware/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/gpu/get_boxes.cpp:6:
/usr/include/stdlib.h:774:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/6/bits/stl_algo.h:59:0,
                 from /usr/include/c++/6/algorithm:62,
                 from /home/strax90/Autoware/ros/src/computing/perception/detection/vision_detector/libs/dpm_ttic/gpu/get_boxes.cpp:6:
/usr/include/c++/6/cstdlib:180:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/6/cstdlib:172:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/build.make:566: recipe for target 'computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/get_boxes.cpp.o' failed
make[2]: *** [computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/gpu/get_boxes.cpp.o] Error 1
CMakeFiles/Makefile2:15275: recipe for target 'computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/all' failed
make[1]: *** [computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/libdpm_ttic.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 73%] Building CXX object util/packages/autoware_rviz_plugins/CMakeFiles/autoware_rviz_plugins.dir/autoware_rviz_plugins_automoc.cpp.o
[ 73%] Linking CXX shared library /home/strax90/Autoware/ros/devel/lib/libautoware_rviz_plugins.so
[ 73%] Built target autoware_rviz_plugins
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

## Specifications

  - Ubuntu version: 16.04.
  - ROS version: Kinetic
  - Autoware branch: Master

Anyone has an idea what is wrong?

Thank you

Strahinja

wmpang, 2018-10-10:
Did you do "rosdep" to clear out all dependency?

amc-nu, 2018-10-10:
@4n00815 seems you're not using Ubuntu 16.04 standard compiler (gcc 5.4). 
Try downgrading, otherwise you'll be required to fix this kind of errors.

Please, reopen if you need extra help.

closed 2018-10-10 by amc-nu

4n00815, 2018-10-10:
@wmpang I've done that before catkin_make
@amc-nu You are right, I was using version 6.4. When i try to compile with version 5.5 I got following error:

[  1%] Building CXX object sensing/drivers/lidar/packages/hokuyo/CMakeFiles/hokuyo_3d.dir/nodes/hokuyo_3d/hokuyo_3d.cpp.o
92 errors detected in the compilation of "/tmp/tmpxft_00000f82_00000000-6_NormalDistributionsTransform.cpp1.ii".
[  1%] Built target adis16470_node
Scanning dependencies of target image_rectifier
CMake Error at ndt_gpu_generated_NormalDistributionsTransform.cu.o.cmake:266 (message):
  Error generating file
  /home/strax90/Autoware/ros/build/computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/./ndt_gpu_generated_NormalDistributionsTransform.cu.o

computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/build.make:77: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o' failed
make[2]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o] Error 1
CMakeFiles/Makefile2:2160: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all' failed
make[1]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Building CXX object sensing/filters/packages/image_processor/CMakeFiles/image_rectifier.dir/nodes/image_rectifier/image_rectifier_node.cpp.o
[  1%] Linking CXX executable /home/strax90/Autoware/ros/devel/lib/vectacam/vectacam_node
[  1%] Linking CXX executable /home/strax90/Autoware/ros/devel/lib/fake_drivers/fake_camera
[  1%] Built target vectacam_node
[  1%] Linking CXX shared library /home/strax90/Autoware/ros/devel/lib/libfastvirtualscan.so
[  1%] Built target fake_camera
[  1%] Built target fastvirtualscan
[  1%] Built target pcl_omp_registration
[  1%] Linking CXX executable /home/strax90/Autoware/ros/devel/lib/image_processor/image_rectifier
[  1%] Built target image_rectifier
[  1%] Linking CXX executable /home/strax90/Autoware/ros/devel/lib/garmin/garmin_gps_18x_lvc
[  1%] Built target garmin_gps_18x_lvc
[  1%] Linking CXX executable /home/strax90/Autoware/ros/devel/lib/hokuyo/hokuyo_3d
[  1%] Built target hokuyo_3d
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

Version 5.4 was upgraded to 5.5 during ROS installation for some reason, and I couldn't avoid that.

Thank you for your replies.

4n00815, 2018-10-11:
Solved by downgrading GCC version to 5.4. 

Thank you everyone for help!