autowarefoundation/autoware #1591: [Feature] Improve A* planner for freespace and avoidance

aohsato, 2018-10-14:
# New Feature
Currently, we have `astar_navi` for free space planning and `obstacle_avoid` for avoidance behavior on waypoints, however, they have each code of Hybrid A* search algorithms and don't work effectively.
So, we need to merge their codes and create the library, then improve its performance.

kennedywai, 2018-10-18:
Hello @aohsato, thanks for trying to imporve astar_navi! BTW I have a few questions about how this planner works:

1) Which branch should I use to perform the astar planner + obstacle avoidance?

2) Is it not required to use either vector map and waypoint saver methods?

3) Is it the finall output that we set up a goal wherever on the free areas on the grid map and it will publish a global path and a velocity to the robot until it reaches the goal? Just like a normal 2D slam on ROS Navigation stack, we dont need to create a lane on vector map.

4) I tried to follow the instructions above but I could not get it working, the astar path wasn't out on Rviz, is that ok that you could upload your testing video on Youtube and share the link here?

Thank you!

aohsato, 2018-10-31:
@kennedywai Thanks for your question and sorry for late!

1. By this branch, you can use `astar_navi` for freespace planning and `astar_avoid` for obstacle avoidance on your waypoints. Could you please help me to test these?

1. You don't need vector map for both nodes. `astar_navi` can generate waypoints so you don't need waypoint saver methods, but `astar_avoid` needs your waypoints generated in some way.

1. That's right. `astarr_navi` needs just grid map and goal pose. It publishes planned waypoints until reaching per constant at a constant cycle, decided by`update_rate` parameter.

kennedywai, 2018-11-03:
@aohsato Thanks for replying me! I will be happy to test this function but I would like to know if this following steps are correct for testing:

1) NDT matching
2) lidar_eclidean_cluster_detector
3) points_to_costmap
4) astar_navi node
5) astar_avoid node
6) velocity_set node
7) twist_filter node
8) Pure pursuit node(waypoint mode)

Correct me if I am wrong about the procedures. Also the waypoints from astar_avoid and astar_navi are the same?

aohsato, 2018-11-07:
@kennedywai And you need `grid_map_filter` and `vel_pose_connect` additionaly.

aohsato, 2018-11-07:
I updated my PR and created README for astar_navi/avoid.

https://github.com/CPFL/Autoware/blob/efe680b27facdc5bd4b228a2afbcc79cf194a769/ros/src/computing/planning/mission/packages/freespace_planner/README.md

https://github.com/CPFL/Autoware/blob/efe680b27facdc5bd4b228a2afbcc79cf194a769/ros/src/computing/planning/motion/packages/waypoint_planner/README.md

kennedywai, 2018-11-17:
@aohsato Hi, sorry for this super late reply, I finally had time to test this feature recently. Btw, I have questions about `astar_avoid` after spending some time testing it:

1. When use `lidar_fake_perception` node for testing `astar_avoid` with `wf_simulator`, I know what exactly this node is publishing but I don't see any outputs and topics from this node being published to `astar_avoid` and `velocity_set`. And of course when I place the fake points in front of sim_base_link, it will not stop.
 
2. I dont have a vector map including `wayarea` and I dont plan on using vector map with `astar_avoid` because this is why I decided to use `astar_avoid` or `astar_navi`. So my question is do we have to build a global 2D grid map first like using 2D lidar with gmapping? Because now I can only convert my "/point_no_groud" or "/point_lanes" topics to "/realtime_cost_map" to test `astar_avoid` and `astar_navi` nodes.

Nodes that I am using for testing:

- NDT matching
- vel_pose_connect
- lidar_fake_perception
- compare_map_filter
- lidar_eclidean_cluster_detector
- points_to_costmap(/realtime_cost_map)
- astar_navi node
- astar_avoid node
- velocity_set node
- twist_filter node
- Pure pursuit node(waypoint mode)

aohsato, 2019-03-14:
@kennedywai Additionaly, you can use `costmap_generator` by @k0suke-murakami, for generating better costmap based on detected bounding boxes, detected convex hulls, lidar point clouds, and wayarea in vecto map. Plz check.

closed 2019-03-14 by aohsato