autowarefoundation/autoware #1598: [Question] Stability of  lidar_euclidean_cluster_detect

wmpang, 2018-10-15:
I tried the lidar_euclidean_cluster_detect to obtain bounding boxes of the clustered point clouds. However, it seems that even there is not much changes in the point cloud, the bounding boxes kept changing significantly. Is there any idea on this problem? Thanks.

hakuturu583, 2018-12-06:
lidar_euclidean_cluster_detect execute euclidean clustering and make bounding box for each clusters.
lidar_euclidean_cluster_detect execute euclidean clustering /poins_no_ground which comes from ray_gound_filter node.
So, please tune parameters for lidar_euclidean_cluster_detect or ray_ground_filter.

zlg9folira, 2020-04-01:
I am trying to run the `lidar_euclidean_cluster_detect`. In Rviz, I am able to see the colors associated to different clusters, however, I am not able to visualize `BoundingBoxArray` according to the steps done in this autoware [video](https://www.youtube.com/watch?v=Tma2DKMxt4Y). Actually, I don't have the bounding boxes published as a topic. Do I miss any configuration inside autoware or what could be the issue?

InterKnight, 2021-05-26:
> I am trying to run the `lidar_euclidean_cluster_detect`. In Rviz, I am able to see the colors associated to different clusters, however, I am not able to visualize `BoundingBoxArray` according to [...]

Hi, I met the same problem. Have you fix it? Thank you !

JWhitleyWork, 2021-05-27:
@InterKnight Please do not bump very old threads. Per our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines), please create a new issue at [ROS Answers](https://answers.ros.org/) with the `autoware` tag.

closed 2021-05-27 by JWhitleyWork