autowarefoundation/autoware #1604: [Bug] lidar_euclidean_cluster_detect fails in pcl::PCA

aohsato, 2018-10-15:
# Bug
`lidar_euclidean_cluster_detect` dies by failing the initialization of pcl::PCA by not enough the number of input points.

## Actual Behavior
```
[ INFO] [1539635915.112029164, 1504784105.808178388]: output pictograms topic: cluster_id
[ INFO] [1539635915.113312032, 1504784105.808178388]: euclidean_cluster > Setting points node to points_no_ground
[ INFO] [1539635915.113833537, 1504784105.808178388]: Euclidean Clustering: Difference of Normals will not be used.
[ INFO] [1539635915.114813272, 1504784105.808178388]: downsample_cloud: 0
[ INFO] [1539635915.115827821, 1504784105.808178388]: remove_ground: 0
[ INFO] [1539635915.116996161, 1504784105.808178388]: leaf_size: 0.100000
[ INFO] [1539635915.118040042, 1504784105.808178388]: cluster_size_min 1
[ INFO] [1539635915.120016066, 1504784105.808178388]: cluster_size_max: 100000
[ INFO] [1539635915.121500387, 1504784105.808178388]: pose_estimation: 0
[ INFO] [1539635915.122457525, 1504784105.808178388]: clip_min_height: -5.000000
[ INFO] [1539635915.123597625, 1504784105.808178388]: clip_max_height: 5.000000
[ INFO] [1539635915.124757906, 1504784105.808178388]: keep_lanes: 0
[ INFO] [1539635915.126024989, 1504784105.808178388]: keep_lane_left_distance: 5.000000
[ INFO] [1539635915.128633350, 1504784105.808178388]: keep_lane_right_distance: 5.000000
[ INFO] [1539635915.130116535, 1504784105.808178388]: max_boundingbox_side: 10.000000
[ INFO] [1539635915.131377155, 1504784105.808178388]: cluster_merge_threshold: 3.700000
[ INFO] [1539635915.133857033, 1504784105.808178388]: output_frame: velodyne
[ INFO] [1539635915.136488935, 1504784105.808178388]: use_vector_map: 0
[ INFO] [1539635915.141899972, 1504784105.808178388]: vectormap_frame: map
[ INFO] [1539635915.143077610, 1504784105.808178388]: remove_points_upto: 0.120000
[ INFO] [1539635915.144130352, 1504784105.808178388]: clustering_distance: 0.750000
[ INFO] [1539635915.145232062, 1504784105.808178388]: use_gpu: 0
[ INFO] [1539635915.146202534, 1504784105.808178388]: use_multiple_thres: 0
[ INFO] [1539635915.147177671, 1504784105.808178388]: clustering_distances: [0.5,1.1,1.6,2.1,2.6]
[ INFO] [1539635915.148568599, 1504784105.808178388]: clustering_ranges: [15,30,45,60]
[ INFO] [1539635915.153830755, 1504784105.808178388]: wayarea_gridmap_topic: grid_map_wayarea
[ INFO] [1539635915.154842094, 1504784105.808178388]: wayarea_gridmap_layer: wayarea
[ INFO] [1539635915.155821042, 1504784105.808178388]: wayarea_no_road_value: 255s
terminate called after throwing an instance of 'pcl::InitFailedException'
what():  bool pcl::PCA<PointT>::initCompute() [with PointT = pcl::PointXYZ] in /usr/include/pcl-1.7/pcl/common/impl/pca.hpp @ 73 : [pcl::PCA::initCompute] number of points < 3
[lidar_euclidean_cluster_detect-1] process has died [pid 1946, exit code -6, cmd /home/aohsato/workspace/projects/autoware/Autoware/ros/devel/lib/lidar_euclidean_cluster_detect/lidar_euclidean_cluster_detect __name:=lidar_euclidean_cluster_detect __log:=/home/aohsato/.ros/log/1a3c43e2-d0b9-11e8-92d0-f85971c90d06/lidar_euclidean_cluster_detect-1.log].
log file: /home/aohsato/.ros/log/1a3c43e2-d0b9-11e8-92d0-f85971c90d06/lidar_euclidean_cluster_detect-1*.log
```

## Steps to Reproduce the Problem

  1. Launch Autoware and publish input pointcloud topic, e.g. `points_no_ground`
  1. Launch `lidar_euclidean_cluster_detect`

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: develop

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

aohsato, 2018-10-15:
@amc-nu Additionally, I used CPU implementation under no GPU enviroment.

amc-nu, 2018-10-16:
Please select minimum number of points of cluster >3 in the UI

aohsato, 2018-10-16:
@amc-nu I fixed in code. Please check.

closed 2018-10-16 by aohsato