autowarefoundation/autoware #1615: Error in launching Rviz from Autoware Docker

KetkiChaudhary, 2018-10-17:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
Unable to run Rviz from the Runtime Manager. 

So, I have installed Autoware using the docker method as described in:
https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker

Now, when I try to run the demo ( as per : https://github.com/CPFL/Autoware-Manuals/blob/master/en/Autoware_QuickStart_v1.1.pdf ), I get stuck at the step 3.2 where I try and launch Rviz from the runtime manager.
I get the following error:

[ INFO] [1539583851.181537990]: rviz version 1.12.16
[ INFO] [1539583851.181578002]: compiled against Qt version 5.5.1
[ INFO] [1539583851.181587881]: compiled against OGRE version 1.9.0 (Ghadamon)
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/home/autoware/Autoware/ros/src/util/packages/runtime_manager/../../../.config/rviz/cmd.sh: line 44:  2362 Segmentation fault      (core dumped) rosrun rviz rviz

Note: I have both Docker CE and Nvidia docker installed and am using Nvidia GTX 1080.

## Expected Behavior
Rviz should launch successfully.

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic
  - Autoware branch: master, v1.7

ajay1606, 2018-11-02:
I am also having the problem with Launching RVIZ in autoware Docker. 
I am using NVIDIA GTX 1070 graphics card.  
Ubuntu version: 16.04
ROS version: kinetic
Autoware branch: master, v1.7

could anyone help to fix this issue.

RomanDmowski, 2018-11-03:
Did you install nvidia-docker1 or nvidia-docker2? Nvidia-docker2 requires libglvnd (GL Vendor-Neutral Dispatch library) to be installed inside the image to get OpenGL calls working correctly. Probably Autoware docker doesn't have OpenGL installed inside so it can't work with nvidia-docker2. See details here: http://wiki.ros.org/docker/Tutorials/Hardware%20Acceleration

ajay1606, 2018-11-05:
Thanks for your reply,
Yes you are right i am using NVIDIA-Docker2:
But,  I didn't understand one thing, Currently, I have autoware docker file. 
According to the link, Do I edit existing docker file or create a new file?

RomanDmowski, 2018-11-05:
I didn't try this solution (to rebuild Autoware docker).
There are other options:
- try to use nvidia-docker1   or
- try to use nvidia-docker2 with  OpenGL inside -  https://hub.docker.com/r/nvidia/opengl/ 

I tried a lot of solution and after a few days I decided not to use docker images. I built Autoware from source. And it works.

mdemirst, 2018-11-05:
I had the same issue, too. It is related to using nvidia-docker2. There is an open PR to solve this problem. 

https://github.com/CPFL/Autoware/pull/1416

Steps: Replace build.sh and run.sh files (+ created Dockerfile.nvidia-docker-v2) with the ones in the PR. or just merge PR into your branch.

ajay1606, 2018-11-06:
> I had the same issue, too. It is related to using nvidia-docker2. There is an open PR to solve this problem.
> 
> #1416
> 
> Steps: Replace build.sh and run.sh files (+ created Dockerfile.nvidia-docker-v2) with the ones in the PR. or just merge PR into your branch.

Thank you so much for your kind response. Much appreciated. it worked for me.

ajay1606, 2018-11-13:
Hello Yugli, did you tried command with $sudo ?? this seems to access permission, So just try adding sudo

yuqli, 2018-11-13:
@ajay1606 Thanks for the reply. I did used sudo. This is the command I run:
```
sudo sh build.sh kinetic nvidia
```
It seems there are actually two errors:
- [Errno 13] Permission denied: '/CMakeLists.txt'/bin/bash: ./catkin_make_release: No such file or directory    
- pull access denied for autoware-kinetic, repository does not exist or may require 'docker login'

The second error I later found out is[ this issue: ](https://github.com/CPFL/Autoware/issues/1537#issuecomment-436133569) the error is also "kinetic image cannot find". I also did not change the autoware version tag as suggested in t[he official installation guide](https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker)

Modifying the run.sh file in [this pull reques](https://github.com/CPFL/Autoware/pull/1416/files#diff-7fd7fd06d35e8cc9b2e975d5c3fa374a)t, change the docker build command to `docker build --build-arg FROM_ARG=autoware/autoware:latest-kinetic -t autoware-$1-$2 -f Dockerfile.nvidia-docker-v2 .` solves the second issue.

I am still looking into the first issue.

yuqli, 2018-11-13:
Still have not resolve the first error (Permission denied) I had, but rviz is running now in docker. Thanks!

ajay1606, 2018-11-13:
Hello Yuqli,
I understood your issue, Seems you have not downloaded the image.
Even I also got same errors in the first time. Later I followed the link below to download the image.
http://ai4sig.org/2018/07/docker-for-autoware/
Try this steps to download the image.
 Good luck :)

yaozhai, 2019-01-13:
> Still have not resolve the first error (Permission denied) I had, but rviz is running now in docker. Thanks!

Exactly the same issue here.

ub-mri, 2019-02-17:
Here you can find the list of all necessary steps to install Autoware with version 2.0 of NVIDIA Docker Plugin and Docker Image based on the image prepared by LG Electronics America R&D Center:   https://icave2.cse.buffalo.edu/AutowareInstallation.htm

RomanDmowski, 2019-02-17:
Here you can find the list of all necessary steps to install Autoware with version 2.0 of NVIDIA Docker Plugin and Docker Image based on the image prepared by LG Electronics America R&D Center:   https://icave2.cse.buffalo.edu/AutowareInstallation.htm

amc-nu, 2019-02-23:
As for release 1.10. **Nvidia Docker v1** is the *preferred method*. 
Like some of you have kindly pointed out, the current Autoware image does not contain OpenGL files, therefore will cause errors when using *nvidia-docker2*
Version 1.11 will address this issue, this is being worked out in this PR: If you are interested, you try to pull and test: https://github.com/CPFL/Autoware/pull/1946
As a side note, in our tests, **nvidia-docker2** works well only on Optimus laptops ,where OpenGL is not installed.
Thank you everyone for all the provided solutions.

closed 2019-02-23 by amc-nu