autowarefoundation/autoware #1623: How to realize lidar fake perception

betsyweilin, 2018-10-19:
# Bug
## Expected Behavior
Expect realize the function displayed in :
https://github.com/CPFL/Autoware/tree/88a1f088eba5878d88e95ecb88cd9389a4bad4b3/ros/src/computing/perception/detection/lidar_detector/packages/lidar_fake_perception

## Actual Behavior
I can not realize this function

## Steps to Reproduce the Problem

  1. run ros
  2. select  waypoint follower  reference items
  3. select  computing/lidar_detector/lidar_fake perception
  4. open rviz, set 2d pose estimate, then the car can run follow the waypoint
  5.  I can publish  /move_base_simple/goal  /fake_twist from rviz , but I don't know how to publish /detected_objects and /points_raw, do we need to play a rosbag ? Any details instruction for that?
  6.  As this node publish /fake_objects and /fake_points, I can not find which module use them, any details about that?
  Thanks

## Specifications

  - Ubuntu version:16.04
  - ROS version: kinetic
  - Autoware branch: 1.8

# New Feature

aohsato, 2018-11-01:
@betsyweilin Did you already checked `/fake_objects` and `/fake_points`? You don't need rosbag or sensor data for publishing.
I think this is TF frame connection error, so please set `output frame`.

BoxTissue, 2018-11-28:
@aohsato 
I am facing the same situation.
I confirmed `/fake_objects` and `/fake_points` topics were published from Lider_fake_perception node.
But there is no node which subscribe `/fake_objects` and `/fake_points`.

I want to execute A* search in `velocity_set` with demo ROSBAG data. But it doesn't include obstacle data.

Which setting should be modified?

closed 2020-05-29 by gbiggs