autowarefoundation/autoware #1625: IMM UKF PDA TRACK

mouryaraj, 2018-10-20:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
As per my project i need to track the dynamic objects and assign them an track ids and it should be consistent

## Actual Behavior
The IMM-UKF-PDA track was not able to detect or track the dynamic objects constantly.  

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS version:
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

wmpang, 2018-10-22:
I found similar problem as described.

mouryaraj, 2018-10-22:
So you solved that issue bro ?

wmpang, 2018-10-23:
No, I still haven't solved the issue and seems other lidar trackers won't work too.

mouryaraj, 2018-10-24:
yeah bro .. have some issues trying to solve it if solve any one them let  me know ..

cirpue49, 2018-10-24:
Hi, 
Could you describe more about your bug?
You see some tracking results? or you cannot see any results of tracking?

mouryaraj, 2018-10-25:
Hello, 

I am able to see the tracking results.
But Id’s which we are assigning for the dynamic or moving objects are not consistent they are keep on changing their ids .

cirpue49, 2018-10-26:
It is not bug.
If your sensor input data is unstable, it is difficult to track objects constantly.

We would appreciate if you send a pull request for modifying the quality of detector or tracker!

closed 2018-10-26 by cirpue49