autowarefoundation/autoware #1628: General Guidelines for Architecture Design

kfunaoka, 2018-10-21:
# General Guidelines for Architecture Design
## Definitions
**Deterministic**: Same output assuming the same sensor data input.

## Guidelines
1. The software is organized in modules called nodes which are represented as ROS nodes in code
1. Every node must handle one specific task (separation of concerns).
1. The coupling between nodes should be as low as possible.
1. Nodes are only allowed to communicate with each other over ROS
1. MSG files are the interface specification of the nodes. Therefore their design and documentation must be done with great care and foresight.
1. Nodes are not allowed to share threads to reduce coupling on a resource level
1. Nodes must not depend on implementation details of other nodes.
1. Every node has one main thread which runs a while loop.
1. Sequential parallelization inside a node is not allowed. This is done on the ROS level.
1. Parallel parallelization must be done using e.g. TBB or OpenMP
1. Before processing data everything needs to be transformed into the same target frame and target time. Every node must have exactly one triggering topic which triggers a new iteration of the while loop on data received. See [link](https://roscon.ros.org/2018/presentations/ROSCon2018_deterministic_asynchronous_message_driven_task_execution.pdf)
1. Use the timestamp of the triggering data.
1. Deterministically calculate the target time based on the timestamp of the triggering data
1. The access of the system clock is not allowed (std::chrono::system_clock etc). In other words the system clock must not influence the behavior of the system. 
1. Sleeping based on duration or time is not allowed (only waiting for external triggers).
1. Processes are not used to modularize the software. Nodes are used for this purpose.

## Guidelines C++
1. OOP and RAII paradigms to be used strictly through the code.
1. Exceptions must be used. Return values are not allowed.
1. Non-const reference members are not allowed.
1. The use of points must be technically justified and this justification must be documented.
1. Use references over pointers.
1. Use raw pointers to indicate non-ownership pointers and smart pointers for ownership pointers.
1. Prefer unique pointers over shared pointers.

mc-android-developer, 2019-04-02:
> Nodes are only allowed to communicate with each other over ROS

What if ROS is not capable to convey large messages (e.g. raw images for multiple cameras) with requited frequency?

gbiggs, 2019-04-03:
Fortunately ROS does have that capability.

closed 2020-05-29 by gbiggs