autowarefoundation/autoware #1647: error at last step in "source build"

dongfang-steven-yang, 2018-10-24:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

I was following the tutorial to build Autoware from source: https://github.com/CPFL/Autoware/wiki/Source-Build

At the last step, when execute "./catkin_make_release", it should successfully compile. However, it returns the following error. I have posted the error message in the next section. 

I've installed all prerequisites mentioned in the tutorial. 

## Actual Behavior

steven@steven-B85M-D3H:~/Autoware/ros$ ./catkin_make_release
Base path: /home/steven/Autoware/ros
Source space: /home/steven/Autoware/ros/src
Build space: /home/steven/Autoware/ros/build
Devel space: /home/steven/Autoware/ros/devel
Install space: /home/steven/Autoware/ros/install
####
#### Running command: "cmake /home/steven/Autoware/ros/src -DCATKIN_DEVEL_PREFIX=/home/steven/Autoware/ros/devel -DCMAKE_INSTALL_PREFIX=/home/steven/Autoware/ros/install -G Unix Makefiles" in "/home/steven/Autoware/ros/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/steven/Autoware/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /home/steven/anaconda3/bin/python (found version "3.7") 
-- Using PYTHON_EXECUTABLE: /home/steven/anaconda3/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/steven/Autoware/ros/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /home/steven/anaconda3/bin/nosetests
ImportError: "from catkin_pkg.package import parse_package" failed: No module named 'catkin_pkg'
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/steven/anaconda3/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/steven/Autoware/ros/build/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:163 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/steven/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/steven/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

## Steps to Reproduce the Problem

  1. follow the tutorial: https://github.com/CPFL/Autoware/wiki/Source-Build
  1. until last step when execute: "./catkin_make_release"
  1.

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic
  - Autoware branch: master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

sgermanserrano, 2018-10-25:
@dongfang-steven-yang the issue is due to you anaconda installation
`export PATH=”$PATH:/home/$USER/anaconda3/bin”` before `./catkin_make_release` should do the trick

dongfang-steven-yang, 2018-10-26:
> @dongfang-steven-yang the issue is due to you anaconda installation
> `export PATH=”$PATH:/home/$USER/anaconda3/bin”` before `./catkin_make_release` should do the trick

Hey, thanks for replying me. Now the previous problem solved. However, there is another problem that I think is related to a module "enum" in python. I searched the Internet and it seems like there is a version conflict? Do you have any idea about this?

Thanks,

Please see the following error messages:

steven@steven-B85M-D3H:~/Autoware/ros$ ./catkin_make_release
Base path: /home/steven/Autoware/ros
Source space: /home/steven/Autoware/ros/src
Build space: /home/steven/Autoware/ros/build
Devel space: /home/steven/Autoware/ros/devel
Install space: /home/steven/Autoware/ros/install
####
#### Running command: "cmake /home/steven/Autoware/ros/src -DCATKIN_DEVEL_PREFIX=/home/steven/Autoware/ros/devel -DCMAKE_INSTALL_PREFIX=/home/steven/Autoware/ros/install -G Unix Makefiles" in "/home/steven/Autoware/ros/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/steven/Autoware/ros/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /home/steven/anaconda3/bin/python (found version "3.7") 
-- Using PYTHON_EXECUTABLE: /home/steven/anaconda3/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/steven/Autoware/ros/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /home/steven/anaconda3/bin/nosetests
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py", line 38, in <module>
    import argparse
  File "/home/steven/anaconda3/lib/python3.7/argparse.py", line 87, in <module>
    import re as _re
  File "/home/steven/anaconda3/lib/python3.7/re.py", line 143, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/steven/anaconda3/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/steven/Autoware/ros/build/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:163 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:52 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/steven/Autoware/ros/build/CMakeFiles/CMakeOutput.log".
See also "/home/steven/Autoware/ros/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

dongfang-steven-yang, 2018-11-08:
Solved by reinstalling Ubuntu.

closed 2018-11-08 by dongfang-steven-yang

AI-liu, 2019-03-02:
in Terminal  :
export PATH=/usr/bin:$PATH