autowarefoundation/autoware #1655: install single package

kargarisaac, 2018-10-26:
I want to use only one package from autoware. when I use catkin_make in catkin_ws I give the error about not finding autoware_build_flags. It's required in CMakeLists.txt. How can I use only one single package? how feed data (pcd or bag?) to topics and see the results. any help or tutorial would be great.
thanks 

# Bug
## Expected Behavior
catkin_make one single package (points_preprocessor)

## Actual Behavior

Base path: /home/isaac/catkin_ws
Source space: /home/isaac/catkin_ws/src
Build space: /home/isaac/catkin_ws/build
Devel space: /home/isaac/catkin_ws/devel
Install space: /home/isaac/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/isaac/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/isaac/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /home/isaac/miniconda3/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/isaac/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - points_preprocessor
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'points_preprocessor'
-- ==> add_subdirectory(points_preprocessor)
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "autoware_build_flags" with any of the following names:

    autoware_build_flagsConfig.cmake
    autoware_build_flags-config.cmake

  Add the installation prefix of "autoware_build_flags" to CMAKE_PREFIX_PATH
  or set "autoware_build_flags_DIR" to a directory containing one of the
  above files.  If "autoware_build_flags" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  points_preprocessor/CMakeLists.txt:4 (find_package)

-- Could not find the required component 'autoware_build_flags'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "autoware_build_flags" with any of the following names:

    autoware_build_flagsConfig.cmake
    autoware_build_flags-config.cmake

  Add the installation prefix of "autoware_build_flags" to CMAKE_PREFIX_PATH
  or set "autoware_build_flags_DIR" to a directory containing one of the
  above files.  If "autoware_build_flags" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  points_preprocessor/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/isaac/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/isaac/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

## Steps to Reproduce the Problem

  1. copy point_preprocessor package into catkin_ws/src 
  2. cd ~/catkin_ws
  3. catkin_make 

## Specifications

  - Ubuntu version:16.04
  - ROS version: kinetic 
  - Autoware branch: 

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

sgermanserrano, 2018-10-26:
@kargarisaac the _points_preprocessor_ package depends on:
- autoware_build_flags
- velodyne_pointcloud (which depends on the other velodyne packages)
- autoware_msgs

so you'll need the 3 of them in your workspace.
Regarding rosbags use, I'd suggest going through the [ROS tutorials](http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data)

kargarisaac, 2018-10-26:
@sgermanserrano 
Thank you for your reply. how can I know package dependencies? I want to install other packages too.

sgermanserrano, 2018-10-26:
have a look at the CMakeLists of the package, you should go over the rest of the ROS tutorials to get acquainted with the ROS basics

kargarisaac, 2018-10-26:
thanks @sgermanserrano . 
I successfully built the packages but cannot find them when use rosrun. (based on tutorial)

####
#### Running command: "make cmake_check_build_system" in "/home/isaac/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/isaac/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodynePacket
[  0%] Built target _velodyne_msgs_generate_messages_check_deps_VelodyneScan
[  0%] Built target _autoware_msgs_generate_messages_check_deps_centroids
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ValueSet
[  0%] Built target _autoware_msgs_generate_messages_check_deps_lane
[  0%] Built target _autoware_msgs_generate_messages_check_deps_RemoteCmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_adjust_xy
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ControlCommand
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigPlannerSelector
[  0%] Built target _autoware_msgs_generate_messages_check_deps_CloudClusterArray
[  0%] Built target _autoware_msgs_generate_messages_check_deps_WaypointState
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ImageLaneObjects
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigLaneStop
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigWaypointFollower
[  0%] Built target _autoware_msgs_generate_messages_check_deps_DetectedObjectArray
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigLatticeVelocitySet
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigDecisionMaker
[  0%] Built target _autoware_msgs_generate_messages_check_deps_accel_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigWaypointLoader
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigRandomFilter
[  0%] Built target _autoware_msgs_generate_messages_check_deps_CANData
[  0%] Built target _autoware_msgs_generate_messages_check_deps_dtlane
[  0%] Built target _autoware_msgs_generate_messages_check_deps_image_rect
[  0%] Built target _autoware_msgs_generate_messages_check_deps_icp_stat
[  0%] Built target _autoware_msgs_generate_messages_check_deps_CloudCluster
[  0%] Built target _autoware_msgs_generate_messages_check_deps_CanInfo
[  0%] Built target _autoware_msgs_generate_messages_check_deps_geometric_rectangle
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigVelocitySet
[  0%] Built target _autoware_msgs_generate_messages_check_deps_brake_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_CANPacket
[  0%] Built target _autoware_msgs_generate_messages_check_deps_indicator_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigLaneSelect
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target _autoware_msgs_generate_messages_check_deps_LaneArray
[  0%] Built target jsk_recognition_msgs_generate_messages_nodejs
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigCarKf
[  0%] Built target _autoware_msgs_generate_messages_check_deps_DetectedObject
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigNdt
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ImageObjects
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ControlCommandStamped
[  0%] Built target _autoware_msgs_generate_messages_check_deps_image_obj_ranged
[  0%] Built target _autoware_msgs_generate_messages_check_deps_image_obj
[  0%] Built target _autoware_msgs_generate_messages_check_deps_Sync_time_monitor
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigVoxelGridFilter
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigPoints2Polygon
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigPedestrianFusion
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigNdtMappingOutput
[  0%] Built target _autoware_msgs_generate_messages_check_deps_lamp_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_steer_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_Signals
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ColorSet
[  0%] Built target _autoware_msgs_generate_messages_check_deps_TrafficLightResult
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ConfigLaneRule
[  0%] Built target _autoware_msgs_generate_messages_check_deps_state_cmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_VehicleCmd
[  0%] Built target _autoware_msgs_generate_messages_check_deps_ScanImage
[... 124 lines cut ...]
[ 21%] Built target autoware_msgs_generate_messages_nodejs
[ 39%] Built target autoware_msgs_generate_messages_py
[ 56%] Built target autoware_msgs_generate_messages_cpp
[ 72%] Built target autoware_msgs_generate_messages_lisp
[ 90%] Built target autoware_msgs_generate_messages_eus
[ 90%] Built target velodyne_msgs_generate_messages
[ 90%] Built target velodyne_input
[ 91%] Built target velodyne_laserscan
[ 91%] Built target autoware_msgs_generate_messages
[ 91%] Built target velodyne_laserscan_node
[ 92%] Built target driver_nodelet
[ 92%] Built target velodyne_node
[ 92%] Built target ringcolors_node
[ 93%] Built target velodyne_rawdata
[ 94%] Built target ringcolors_nodelet
[ 95%] Built target cloud_nodelet
[ 95%] Built target cloud_node
[ 95%] Built target space_filter
[ 97%] Built target transform_nodelet
[ 97%] Built target points_concat_filter
[ 98%] Built target transform_node
[100%] Built target ring_ground_filter
[100%] Built target ray_ground_filter_lib
[100%] Built target cloud_transformer
[100%] Built target ray_ground_filter

sgermanserrano, 2018-10-26:
have you sourced your workspace `setup.bash`?

kargarisaac, 2018-10-26:
yes. I cannot find the packages using rospack list too.

kargarisaac, 2018-10-26:
Solved. first rosdep update and then source setup.bash solved the problem.

closed 2018-11-13 by kargarisaac

Usman-j, 2019-05-09:
> @kargarisaac the _points_preprocessor_ package depends on:
>
> * autoware_build_flags
> * velodyne_pointcloud (which depends on the other velodyne packages)
> * autoware_msgs
>
> so you'll need the 3 of them [...]

Hi @kargarisaac @sgermanserrano ,
Can you kindly guide me on how to add package dependencies to my workspace? I am trying to make a separate package for 'ndt_mapping' (a node in the package 'lidar_localizer').