autowarefoundation/autoware #1663: Mark Indigo Deprecated

kfunaoka, 2018-10-29:
Basing Autoware.AI on ROS2 and using common tools would move the Autoware.Auto and Autoware.AI communities closer, and potentially share more code.

We can start by making it easier to write code for ROS2 now by adding the ros1_bridge as a dependency, just by having it as part of the Autoware distribution, it's easy to write ROS2 nodes that talk with the rest of the current system.

We could mark Indigo as a deprecated target in 1.9 and remove it in a future release, the bridge does not support Indigo.

esteve, 2018-11-09:
Done in https://github.com/CPFL/Autoware/pull/1702

bdholt1, 2019-01-23:
ROS Indigo is no longer supported in 1.10. https://github.com/CPFL/Autoware/releases/tag/1.10.0

closed 2019-01-23 by bdholt1