autowarefoundation/autoware #1665: User Created with Insecure Permissions

harobbins, 2018-10-29:
When creating a user in the docker/generic/Docker.kinetic file you give the user root permissions with the password 'autoware' and nopasswd privileges in the sudoers file. This allows anyone with access to this system to have root permissions without any authentication and should be changed.

hakuturu583, 2018-12-06:
@harobbins I think it is necessary for attackers to login the host machine of the docker image.
This is already insident.
So, I think it is not a problem to write username and password in Dockerfile.
Please tell me what kind of insident are you talking about?
Thanks!!

harobbins, 2018-12-19:
The issue is not that it is written to the Dockerfile, but that the user is created at all. Having a user with root permissions like this is leaving the system open to significant security risk.

filiperinaldi, 2019-02-01:
This is a good point from a product point of view. I'm assuming the Docker support in docker/generic is only a reference for testing and development. A real deployment should not use it.

closed 2020-05-29 by gbiggs