autowarefoundation/autoware #1669: [Feature] diagnostic library for Autoware.

hakuturu583, 2018-10-30:
Now, there is no diagnostic library for Autoware.
It is very dangerous. #1579 

# New Feature
[image]

diagnostic lib contains diag_manager class.
It provides various kinds of functions for diagnostics.

Now, log file format is like below.
```
[2018-09-13T03:25:25.340543] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:25.341312] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.441295] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.541326] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.641427] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.741318] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.841311] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:25.941436] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:26.041322] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:26.141353] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:26.340464] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:27.340491] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:28.241331] : in /ndt_matching: exception was catched
[2018-09-13T03:25:28.241375] : in /ndt_matching: Divided by zero.
[2018-09-13T03:25:28.340549] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:29.340556] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:30.340551] : in /ndt_matching: topic /ndt_matching/data subscribe rate was low (Warn)
[2018-09-13T03:25:30.341377] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:30.441393] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:30.541348] : in /ndt_matching: The input value hogehoge is out of range.
[2018-09-13T03:25:30.641382] : in /ndt_matching: The input value hogehoge is out of range.
```

kfunaoka, 2018-10-30:
@hakuturu583 This issue focus on software error, and hardware error such as sensors is another topic. Is it right?

hakuturu583, 2018-10-30:
Yes.
This library only handles software diagnostics.
Hardware diagnostics will add later.

hakuturu583, 2019-02-04:
this Issue was updated by #1943

closed 2019-02-04 by hakuturu583