autowarefoundation/autoware #1672: Give intrinsic of camera in calibration toolkit

aashi7, 2018-10-30:
I wish to calibrate the camera and velodyne extrinsically. Is there a way to specify known intrinsics (CameraMat and DistCoeffs) of camera and then calibrate for CameraExtrinsicMat?

amc-nu, 2018-10-30:
Yes! You can
1. Open this file
https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/kitti_pkg/kitti_launch/calib/kitti_cam02_sync.yml

1. Copy the contents to a new one

1. Leave the CameraExtrinsicMat just as it is( this will be replaced by the extrinsic calibration )

1. Complete the second part with your values as follows:

```
CameraMat: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [ fx , 0, cx,
           0, fy, cy ,
           0, 0, 1 ]
DistCoeff: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [ r1, r2, r3, r4, r5 ]
ImageSize: [ image_width, image_height ]
```

aashi7, 2018-10-30:
I tried the following:
1) Changed the intrinsic and distortion in yml file.
2) Loaded the updated calibration file in calibration toolkit.
3) After pressing the button 'Calibrate', it changes the intrinsic matrix. Did it only use the above as initialization or it can be kept fixed. Please let me know if I'm doing something wrong. Thanks.

amc-nu, 2018-10-30:
Please use this calibration tool instead
https://github.com/CPFL/Autoware/blob/master/ros/src/sensing/fusion/packages/autoware_camera_lidar_calibrator/README.md#camera-lidar-extrinsic-calibration

closed 2018-11-06 by amc-nu