autowarefoundation/autoware #1677: [Feature] Autoware Vehicle Interface Design

aohsato, 2018-11-01:
# Design
I created the new design and message definitions around the "Actuation Layer".
We're using multiple types of vehicles, ZMP, AS, YMC, and so on. However they have different configurations and I/O, so it's necessary to commonize interface for the low-level layer in Autoware.

Please see this, then let's start to discuss.
https://docs.google.com/presentation/d/154IzTXNisLqc1BJdjKH7DN4D3UGOmnKbnODIHSsTJ5U/edit#slide=id.g468173be5d_0_44

[image]

hakuturu583, 2018-11-02:
I want to use simulator with the same interface with real car.
I think the bridge node between Autoware and simulator should have same interface with actuation node.

hatem-darweesh, 2018-11-02:
how about using similar interface like the attached file. (abstract C++ class)
to develop any low level control node (hardware specific) or even simulation control user can implement the interface. 
I imagine single param in RuntimerManager to select target control platform (zmp car, AS , .. Simulation controller 1, simulation controller 2 .. ). 
a worker node should create the proper implemented object from above. 
Example: 
GeneralCommunication* pComm;
switch(interface_type)
{
case ZMP: 
    pComm = new ZMPcontroller(); 
break 
case AS : 
   pComm = new AScontroller(); 
break;
case WF_Simulator: 
   pComm = new wfController(); 
break;
case MPC_Simulator: 
   pComm = new MPCcontroller();
break;
........
} 

Another solution is to create an interface node and follow the same previous paradigm but with nodes not C++ classes. 
[GeneralCommunication.h.zip](https://github.com/CPFL/Autoware/files/2541369/GeneralCommunication.h.zip)

dejanpan, 2018-11-05:
@aohsato vehicle interface re-design is being discussed and already re-implemented in https://github.com/CPFL/Autoware/issues/1541. 

It seems that your proposal is not really about the vehicle interface but about the Autoware ROS interface to the low level speed and steering controllers (which then in turn interfaces with a DBW module). Please read https://github.com/CPFL/Autoware/issues/1541 and suggest how your proposal fits into https://github.com/CPFL/Autoware/issues/1541. 

OK?
---
Please also note that the way motion planner, low level control and behaviour planner are implemented urgently needs a re-design. 

I did not have time to prepare a concrete proposal yet but it should have the following components (which every serious (e.g. waymo, TRI, Cruise, ..) project has):

1. Global planning => on demand service, interface to apriori map and traffic information, 
2. Behaviour planning => open problem, Darpa urban challenge: rule-based, other companies: search-based method, our favorite: robust POMDP
3. Local planning => some deterministic sampling-based methods (GMT) or form of MPC, pre-recorded path, RRT
4. Control

Inputs to this module:

1. track predictions
2. map
3. pose

YamatoAndo, 2018-11-05:
@aohsato I think that it is better to add the type of drive model (steer, two differential drive, omni etc).

aohsato, 2018-11-05:
@hakuturu583 Yes, of course. Simulators and real vehicles can select command format, like speed/steer or pedal/steer commands, according to the vehicle dynamics.

aohsato, 2018-11-05:
@dejanpan Just as you said, this post title is a little bit wrong so I changed. Thanks much. 
Yes I read that post and I understood "ROS2 Vehicle Node" generated based on "Vehicle Abstraction Class" performs as "Actuation Node" in my figure. Is this correct?

And next, I want to design around Planning and Control too. Of course, I agree. 
So in this issue, I want to decide the architecture after local planner (publishing waypoints including speed and pose, `velocity_set` in current Autoware), especially message definition for Actuation Node. Please help?

aohsato, 2018-11-05:
@YamatoAndo Agreed. Thanks for your suggestion, so I updated my slide.
https://docs.google.com/presentation/d/154IzTXNisLqc1BJdjKH7DN4D3UGOmnKbnODIHSsTJ5U/edit#slide=id.g468173be5d_0_4

kfunaoka, 2018-11-05:
@aohsato I'd like to formalize your proposal into an abstract based on [Overview](https://github.com/CPFL/Autoware/wiki/Overview#motion) and the level of #1501. Since this issue targets the flow after High Level Planner (velocity_planner and trajectory_planner here), I'm creating a diagram that also focus on the flow. This figure is not incomplete. Since I'm not familiar with motion planning, I'd like to discuss with F2F at first :)

[image]

hakuturu583, 2018-11-06:
@aohsato I think it is better to add use_whici_command field to the autoware_msgs/VehicleCmd message.

hakuturu583, 2018-11-06:
@aohsato for example.
```
Header header

autoware_msgs/ModeCmd        mode    # vehicle mode (e.g. auto, manual, emergency
autoware_msgs/BlinkerCmd        blinker    # blinker type (e.g. left, right, hazard
autoware_msgs/ShiftCmd        shift    # shift type (e.g. drive, park, reverse

autoware_msgs/SpeedCmd        speed
    Header        header
    std_msgs/Float32    velocity        # [m/s]
    std_msgs/Float32    acceleration    # [m/s^2]

autoware_msgs/SteerCmd        steer
    Header        header
std_msgs/Float32    angle        # [rad]
    std_msgs/Float32    angular_velocity    # [rad/s]

autoware_msgs/PedalCmd        pedal
    Header        header
    std_msgs/Float32    throttle        # [0.0, 1.0]
    std_msgs/Float32    brake        # [0.0, 1.0]

uint8 use_command
uint8 SPEED_AND_STEER_CMD = 1
uint8 PEDAL_CMD = 2
```

hakuturu583, 2018-11-06:
@aohsato I think simulator will support multi types of command.
So, this field makes us to run simulator easily.

aohsato, 2018-11-06:
@kfunaoka Okey, next I want to modify your diagram based on our F2F discussion so please give me your code?

aohsato, 2018-11-06:
@hakuturu583 I think Actuation nodes for simulator and vehicle have the priority of selecting command type and Autoware doesn't need to decide it. 
If you want to use pedal command, for example, you just have to launch pedal controller in addition to basic motion packages and set up a simulator to use it.

hakuturu583, 2018-11-10:
@aohsato I understand that which command used field was unnecessary when we use Autoware with simulator.
But, when we play rosbag and compare VehicleCmd and VehicleStatus (for turning controllers,etc...) if this field is not exist, we cannot explain which command was used explicitly.

aohsato, 2018-11-13:
Like this.
[image]

`avoid_planner` : local avoidance planner to avoid objects and provide base waypoints, e.g. `astar_avoid`
`target_detector` : detecting objects around base waypoints and publishing them as `TargetObjectArray.msg`, including waypoints index, obstacle velocity, obstacle type, and more.
`velocity_planner` : local speed planner to calculate velocities on waypoints based on `target_detector/targets` 
`motion_controller` : generating `speed_cmd` and `steer_cmd`, e.g. `pure_pursuit`
`motion_filter` : subscribing raw commands and publishing smoothed and safety commands
`pedal_controller` : providing `pedal_cmd` including throttle and brake pedal command based on `speed_cmd` and vehicle dynamics configuration
`command_gateway` : selecting command topics by motion logics or by other interfaces, like joystick or remote client nodes.

aohsato, 2018-11-13:
@hakuturu583 I don't think vehicles select speed/pedal commands dynamically, so it's just a parameter like `rosparam` and you don't need to keep to know it on topics.
But it's a good point how to reproduce them not recorded on rosbag. Actually so, we need the system storing and reproducing parameters.

dejanpan, 2018-11-13:
@aohsato: 

> @dejanpan Just as you said, this post title is a little bit wrong so I changed. Thanks much.
> Yes I read that post and I understood "ROS2 Vehicle Node" generated based on "Vehicle Abstraction Class" performs as "Actuation Node" in my figure. Is this correct?

In #1541 we abstract the vehicle (DBW interface and CAN messages) which makes #1541 be the closest SW to the car interface. In you figure you have Low Level Controller between the Actuation Node and the car which does not make sense (or I do not know what kind of Low Level Controller) are you talking about. But I assume that this Low Level Controller will be written in ROS, will receive desired speed and curvature and it will output to the vehicle throttle/break and acceleration, right?

> And next, I want to design around Planning and Control too. Of course, I agree.
> So in this issue, I want to decide the architecture after local planner (publishing waypoints including speed and [...]

I can help but we need to break the old Autoware architecture. Could we maybe start by defining which inputs/outputs go into these 4 layers and what the respective algorithms in these 4 layers would be?

1. Global planning
2. Behaviour planning
3. Local planning
4. Control

If you want, I will ask Chris for a slightly more detailed architecture tomorrow.

aohsato, 2018-11-13:
@dejanpan I meant `Low Level Controller` is just vehicle interface ECU (like AS pacmod) connected with actuation node, and `pedal_controller` outputs throttle/break pedal command value (like AS LLC).
But it's a little misleading, so I deleted "low level controller" names.

[image]

aohsato, 2018-11-13:
@dejanpan 
> I can help but we need to break the old Autoware architecture. Could we maybe start by defining which inputs/outputs go into these 4 layers and what the respective algorithms in these 4 layers would be?

Agreed. In this post, I want to discuss about `3.Local planning` and `4.Control` only. 
Can I create the issue of planning overview?

kfunaoka, 2019-03-14:
Discussions with @aohsato @gbiggs @mitsudome-r @h-ohta @TakaHoribe @yukkysaito @yn-mrse @s-azumi 

- Delete PedalCmd pedal from VehicleCmd.msg
- Autoware can query vehicle_info to by-wire in the future?
- Built-in safety features can provide their info to Autoware

aohsato, 2019-03-14:
@kfunaoka Thanks. I'll update based on today's discussion.

aohsato, 2019-03-26:
I updated my documentation by following our previous discussion. Finally, I want to decide the ROS interface and message definitions for Actuation layer in this issue. Plz comment?
https://docs.google.com/presentation/d/1CW9R_W0cxq856iRDFn6w5oWQrblGNtzhqnhx1PSwU-w/edit?usp=sharing

kfunaoka, 2019-03-27:
@aohsato I guess motion planning layer will be defined in #1719. The slide defines actuation layer and its input and outputs. The motion planning layer, e.g., motion planner and motion controller, in this slide is just an example. Is my understanding correct?

aohsato, 2019-03-27:
@kfunaoka Right, just example. It's advisable to discuss planning layer in that issue.

closed 2020-05-29 by gbiggs