autowarefoundation/autoware #1682: Release 1.10

esteve, 2018-11-02:
# Announcements

# Security Fixes

# New Features

# Improvements

# Bug Fixes

esteve, 2018-11-02:
@kfunaoka I've created this ticket so that we can have a list of the planned changes for the 1.10 release and decide what should go in and what should be postponed as suggested by @shinpei0208 https://autoware-developer.slack.com/archives/C3C8XRV99/p1541114851079100

kfunaoka, 2018-11-06:
I'd like to also entry https://github.com/CPFL/Autoware/issues/1551 and https://github.com/CPFL/Autoware/issues/1693 once I can make them meaningful :)

kfunaoka, 2018-11-07:
New entry: https://github.com/CPFL/Autoware/pull/1645

kfunaoka, 2018-11-07:
New entry: https://github.com/CPFL/Autoware/issues/1637

kfunaoka, 2018-11-08:
New enty: https://github.com/CPFL/Autoware/issues/1701

esteve, 2018-11-09:
@kfunaoka I've added the tickets to the description

hakuturu583, 2018-12-19:
Sorry for my late.
New entry #1795

cirpue49, 2018-12-20:
Sorry for late. New entry #1685 #1772

esteve, 2018-12-20:
@hakuturu583 thanks for letting us know, but unfortunately it's too late for this release to include #1795. The goal of this ticket is to list all the features and improvements that will go into a release before starting the release, not after. This helps us have a vision of what the priorities are for a given release/milestone and how efforts should be directed.

In any case, it looks like #1795 is a good candidate for Autoware 1.11, we'll add it to the next release.

kfunaoka, 2018-12-21:
Open source community should have a certain amount of discipline to work with a lot of people. I added a procedure into Contribution Rules to declare which version each issue is targeting.
https://github.com/CPFL/Autoware/wiki/Contribution-Rules#development-workflow

@hakuturu583 Sorry for the insufficient announcement. Also I'd like to encourage you to improve Autoware more and more.  For that, let's talk what you are working on within the community :)

@esteve I've also talked with @hakuturu583. Though we can postpone #1795, simulator is a very good tool for autonomous driving, and it's worth for Autoware and users. Since #1795 does not affect the essential part of Autoware, we might be able to decide on importing #1795 to test Autoware functions by challengers. How do you think in this case?

@hakuturu583 Any other comments?

kfunaoka, 2018-12-21:
Sorry, I have to withdraw #1693 and #1551.

esteve, 2019-01-02:
@kfunaoka It was mostly because I think that we should be stating what we're going to work on for the current cycle, so we can anticipate and properly test everything before making a new release. That doesn't mean we can't squeeze in a new feature in the middle of the cycle, but they should have a rationale and give enough time to test it.

Unfortunately this wasn't the case with #1795, it broke the build, doesn't have any tests at all and was merged in a rush.

#1795 is a very valuable contribution and I'd really like it getting merged, but it'd be good to be a bit more disciplined with releases and plan a bit better, it'd help users anticipate any potentially incompatible changes before a release, for example.

hakuturu583, 2019-01-03:
@esteve #1795 have already merged to develop, and travis CI said that the develop branch was not broken.
Please show me the CI log.

esteve, 2019-01-03:
@hakuturu583 we no longer use Travis for CI, and these are the tickets that report that the simulator bridge broke the build:

https://github.com/CPFL/Autoware/issues/1802
https://github.com/CPFL/Autoware/issues/1807

hakuturu583, 2019-01-03:
@esteve OK, I am fixing now.

esteve, 2019-01-03:
@hakuturu583 no worries, it's really not big of a deal :-) I think the work you did in #1795 is really valuable, but let's plan releases in advance so we can have more time to test them more thoroughly.

kfunaoka, 2019-01-05:
@hakuturu583 @esteve That'll make the process more workable :) I'd like to request not only contributers but reviewers to check the release plan of each pull request. I'll create the check list that reviewers reference.

Regarding this release 1.10, I'd like to provide autonomous driving to who has no real cars if it can not be critical. Were the problems fixed or not? I found https://github.com/CPFL/Autoware/pull/1808 was merged.

Finally, I apologize to @zelenkovsky and @hakuturu583. I knew they were working for the release 1.10. But I forgot to request to declare on the issue.
https://github.com/CPFL/Autoware/issues/1724#issuecomment-441248170

kfunaoka, 2019-01-07:
@sgermanserrano @esteve Would you list pull requests that are better to be merged into v1.10? My understanding is:

- Use colcon as the build tool #1704
- Switch to Apache 2 license (develop branch) #1741

sgermanserrano, 2019-01-07:
@kfunaoka @aohsato @cirpue49 as discussed in Slack #motion-planning channel `A*` is better used with `costmap_generator` than with `points2costmap` (which seems older -> can be deprecated and substituted by `costmap_generator`?). 
I'd suggest pushing [Feature] Improve Hybrid A* planner #1594 from v1.10 to v1.11 to:
- Update from nav_msgs::OccupancyGrid to GridMap
- Review and improve tests: #1753 
- Add tests for `freespace_planner` and `astar_avoid`

kfunaoka, 2019-01-07:
@sgermanserrano I heard the discussion from @aohsato. Let's postpone to 1.11!

esteve, 2019-01-07:
@kfunaoka I postponed #1704 because I thought it'd make it for the original 1.10 cutoff, but since it was delayed, I think we can try. I'll keep testing it locally, I'd appreciate your (and @sgermanserrano  's) feedback, since it will have a direct impact on the users.

#1741 should be mergeable, but I need another +1.

kfunaoka, 2019-01-07:
I've updated the status.

- I've added #1824 into the 1.10 list above since localization death is fatal for autonomous driving.
- #1704 is almost ready to merge.
- #1741 is lamost ready to merge but need a fix.

esteve, 2019-01-07:
@kfunaoka I've added #1704 to the list of tickets for this release and merged the PR.

esteve, 2019-01-07:
Once #1741 passes, I'll merge it as well.

sgermanserrano, 2019-01-07:
@isamu-takagi @messi49 do #1637 and #1701 have associated PRs? or are they features on a planning/developing stage?

kfunaoka, 2019-01-08:
@amc-nu @sgermanserrano @esteve How about creating release branch for v1.10, merging #1837 into the release branch, and staring test on the release branch?

- #1837: Should be merged into v1.10.
- #1701: Need more discussion. -> v1.11
- #1645: Might take some time. -> v1.11?

kfunaoka, 2019-01-08:
- #1832: Can be merged into v1.10 to minimize time and effort for test.

amc-nu, 2019-01-08:
Hello @esteve @sgermanserrano and @kfunaoka 

Please don't include #1704 as it is.
I found the following problems:
1. `run` script won't work, as it is.
Please refer to https://github.com/CPFL/Autoware/blob/develop/ros/run#L52
2. Even after updating the script to `install` space. Since currently most of the packages **do not** include the `install` command for launch, and companion files. Only manual `ros run` commands will work.
https://github.com/CPFL/Autoware/blob/develop/ros/src/util/packages/runtime_manager/scripts/computing.yaml
3. Compiling with catkin_make workflow is also broken so it cannot be used as an alternative.

Fixing the above points will require a considerable amount of work. For this reason, I consider appropriate 
1. First to apply #1664 , then 
2. Fix the file installation commands in the CMakeFiles and 
3. Finally include #1704 + fixes in a release.

Otherwise, current users (ourselves included) will face a lot of problems. Please let me know your thoughts on this.

esteve, 2019-01-08:
@amc-nu I've reverted #1704 in https://github.com/CPFL/Autoware/pull/1838, please have a look at it. However, the real problem is that for a long time the Autoware run script assumed that it was using the devel space, which is not a correct workflow in ROS. colcon is stricter about that, there's no devel space at all, only build and install, so if the packages are not properly written and they don't have an install target, they can't be used.

What I find strange is that catkin fails for you. Doesn't catkin_make_release work for you?

kfunaoka, 2019-01-08:
It seems all issues and problems are solved by some directions.
I've created the [release/1.10.0](https://github.com/CPFL/Autoware/commits/release/1.10.0) branched from f77b7849273cffdbb5cc2583527c77d691572c45.
Let me know if something should be done additionally.

Since we changed the build tool from catkin to calcon,
we might need test time a little longer than usual.
How about these target plan?

| Version |  Merged | Release |
|-----------|------------|------------|
| [v1.10](https://github.com/CPFL/Autoware/milestone/4) | Jan. 8th, 2018 (today) | Jan. 17th, 2018 |

Also I'm thinking the docker image for test such as autoware/autoware:1.10.0-kinetic-rc1 will be useful.

esteve, 2019-01-08:
@kfunaoka looks good to me.

kfunaoka, 2019-01-09:
I found the error #1849. I'll check.

kfunaoka, 2019-01-11:
I've uploaded `autoware/autoware:1.10.0-kinetic-rc1` based on b21b663dc4e718e29de96f53efffb766d8af5ebe into dockerhub.

kfunaoka, 2019-01-22:
@esteve @sgermanserrano @amc-nu 
I've created the release note at the headline of this issue.
Let me know if something should be added or modified.
I'll release tomorrow.

kfunaoka, 2019-01-23:
Thanks!

I've updated the wiki page for the new docker scripts.
https://github.com/CPFL/Autoware/wiki/Generic-x86-Docker

bdholt1, 2019-01-23:
Congrats on the release!  Can this issue be closed now?

kfunaoka, 2019-01-23:
Thank you for reminding!

closed 2019-01-23 by kfunaoka