autowarefoundation/autoware #1690: Dataset used in SSD Traffic Light Model

madhavajay, 2018-11-05:
What dataset was used to train the pre-trained Caffe SSD MobileNet model linked here:
https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/detection/trafficlight_recognizer/nodes/region_tlr_ssd/README.md

This file:
http://db3.ertl.jp/autoware/tlr_trained_model/data.tar.bz2

amc-nu, 2019-03-05:
For details please refer to the following publication:
```
@article{HIRABAYASHI201962,
title = "Traffic light recognition using high-definition map features",
journal = "Robotics and Autonomous Systems",
volume = "111",
pages = "62 - 72",
year = "2019",
issn = "0921-8890",
doi = "https://doi.org/10.1016/j.robot.2018.10.004",
url = "http://www.sciencedirect.com/science/article/pii/S0921889018301234"
}
```

closed 2019-03-05 by amc-nu

Wasedarocket, 2020-07-29:
hello, this file link can not be opened, could you be so kind to update an accessible link?

JWhitleyWork, 2020-07-29:
@Wasedarocket Please create a new question on ROS Answers and we will respond there. This question is very old.

mengen100, 2021-01-07:
@Wasedarocket 
I found the updated documentation link, you can download the file from there:
https://github.com/Autoware-AI/core_perception/tree/master/trafficlight_recognizer/nodes/region_tlr_ssd