autowarefoundation/autoware #1697: Add support for Melodic Morenia

esteve, 2018-11-06:
So far the only blocker for Autoware to support Melodic Morenia is the following packages missing:

- [ ] grid_map_msgs https://github.com/ANYbotics/grid_map/issues/171 and https://github.com/ANYbotics/grid_map/pull/179
- [ ] jsk_recognition_msgs https://github.com/jsk-ros-pkg/jsk_recognition/issues/2358
- [ ] jsk_rviz_plugins https://github.com/jsk-ros-pkg/jsk_recognition/issues/2358 
- [ ] jsk_topic_tools https://github.com/jsk-ros-pkg/jsk_recognition/issues/2358

esteve, 2018-11-06:
The JSK packages being missing in Melodic seem to be caused by the PR2 no longer being supported in Melodic and the switch to tf2 in rviz. I'll submit PRs upstream to address that.

esteve, 2018-11-07:
From slack https://autoware-developer.slack.com/archives/C3C8XRV99/p1541593124109000?thread_ts=1541418102.094800&cid=C3C8XRV99

@sgermanserrano said:

the missing packages can be installed from source in Melodic, the ones needed for Autoware:
1 - grid_map:
- grid_map_core
- grid_map_cv
- gird_map_msgs
- grid_map_ros
2 - jsk_common:
- jsk_common
- jsk_topic_tools
3 - jsk_recognition:
-  jsk_recognition_msgs
-  jsk_recognition_utils
4 - jsk_visualization:
- jsk_rviz_plugins
Also:
- The CMakeLists in computing/perception/detection/vision_detector/libs/dpm_ttic needs to be updated to use PCL 1.8 instead of 1.7
- The CMakeLists in util/packages/autoware_bag_tools needs to add tinyxml in the _target_link_libraries_
- There is an error in actuation/vehicles/packages/ymc: _relocation R_X86_64_32 against `.rodata' can not be used when making a PIE object; recompile with -fPIC_ the static libraries might need recompilation

esteve, 2019-01-08:
The following packages depend on JSK packages, some of them are obsole or will be according to https://docs.google.com/spreadsheets/d/1wcMiQJFr3SwHAS9H21UbcmnmMMDvq04Jpa7uGml0MdI/edit

- autoware_can_msgs https://github.com/CPFL/Autoware/pull/1840
- autoware_msgs https://github.com/CPFL/Autoware/pull/1840
- decision_maker
- dp_planner #1812 (ToBeObsolete)
- kitti_box_publisher
- lidar_euclidean_track #1812 (Obsolete)
- lidar_kf_contour_track #1812 (Obsolete)
- lidar_kf_track #1812 (Obsolete)
- lidar_svm_detect #1812 (Obsolete)
- obj_fusion #1812 (Obsolete)
- obj_reproj #1812 (Obsolete)
- op_ros_helpers
- pos_db
- range_vision_fusion

IndefiniteBen, 2019-05-07:
Any idea when this will be fixed? ANYbotics/grid_map#185 [says](https://github.com/ANYbotics/grid_map/pull/185#issuecomment-466540385) that it was last dependency for melodic support with autoware, so the check list in this issue is all complete and can be checked?

Is there an existing branch with these changes, or when will it be moved to release? Thanks

closed 2019-05-31 by amc-nu