autowarefoundation/autoware #1705: Invalid message in quick_start launch file

MarcoCecotti, 2018-11-09:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
In release 1.9.1, launching the file https://github.com/CPFL/Autoware/blob/master/docs/quick_start/my_mission_planning.launch causes an error.

## Expected Behavior
No error.

## Actual Behavior
ERROR: invalid message type: autoware_msgs/ConfigWaypointLoader.
If this is a valid message type, perhaps you need to type 'rosmake autoware_msgs'
[config_waypoint_loader_rostopic-4] process has died [pid 19852, exit code 1, cmd /opt/ros/kinetic/lib/rostopic/rostopic pub -l /config/waypoint_loader autoware_msgs/ConfigWaypointLoader {multi_lane_csv: /home/marco/.autoware/data/path/moriyama_path.txt, replanning_mode: false, velocity_max: 20.0, velocity_min: 4.0, accel_limit: 0.98, decel_limit: 0.98, radius_thresh: 20.0, radius_min: 6.0, resample_mode: true, resample_interval: 1.0, velocity_offset: 4, end_point_offset: 5} __name:=config_waypoint_loader_rostopic __log:=/home/marco/.ros/log/a5e4d100-e42f-11e8-a369-b0359fcde2a4/config_waypoint_loader_rostopic-4.log].
log file: /home/marco/.ros/log/a5e4d100-e42f-11e8-a369-b0359fcde2a4/config_waypoint_loader_rostopic-4*.log

## Steps to Reproduce the Problem

Follow the demo https://github.com/CPFL/Autoware/wiki/Demo.

hakuturu583, 2018-12-06:
@MarcoCecotti Why are you use rosmake??
We only support indigo and kinetic.

closed 2018-12-06 by hakuturu583

MarcoCecotti, 2018-12-13:
Hi @hakuturu583,

I don't understand your comment: [rosmake](http://wiki.ros.org/rosmake) is fully supported in indigo AND kinetic. In addition to this, rosmake doesn't have anything to do with the error.

The error is caused by a change of path: the message definition ConfigWaypointLoader has been moved to the folder autoware_config_msgs. Please see my commit referencing this issue for more information.

Kind regards,

Marco

pedrobagatin, 2019-02-07:
Got same error, did you find out what was the problem?
I ran the "rosmake atoware_msgs" but didn't work.

pedrobagatin, 2019-02-07:
> Hi @hakuturu583,
>
> I don't understand your comment: [rosmake](http://wiki.ros.org/rosmake) is fully supported in indigo AND kinetic. In addition to this, rosmake doesn't have anything to do with the [...]

Is there a way to solve this problem in PC I am running? Can we set correct msgs folders?

MarcoCecotti, 2019-02-07:
Hi @pedrobagatin,

You simply have to replace the string autoware_msgs with autoware_config_msgs in docs/quick_start/my_mission_planning.launch (the launch file that you copied in your message).