autowarefoundation/autoware #1707: [Discussion] Transition to Apache License 2.0

kfunaoka, 2018-11-12:
Apache License 2.0 might be preferable for Autoware, e.g., 3. Grant of Patent License.

Let's discuss.

kfunaoka, 2018-11-12:
> We'd have to ask contributors for permission to relicense their contributions.
> At this stage, it might involve a lot of work

It seems difficult.

asimay, 2018-11-26:
Agree.

andoh104, 2018-11-26:
Agree.

esteve, 2018-11-26:
@ChenxiTU I saw you ticked the checkbox, but unfortunately I'll need you to explicitly approve the license change, just type +1 and I'll take care of checking that everyone has given their approval.Thanks.

esteve, 2018-11-27:
@hiro-ya-iv @shinpei0208 do you mind adding a comment with "+1" so that we have a trail of who approved the change? Reacting with a thumb up on the ticket unfortunately leaves no trace and can't be linked. Thanks.

niwasaki, 2018-11-28:
> List of contributors:
> - [ ] @KenYN
> - [ ] @KenjiroYamada
> - [ ] @Mani-Sadhasivam
> - [ ] @MasatoTange
> - [ ] @TomohitoAndo
> - [ ] @Tomoya-Sato
> - [ ] @YamatoAndo
> - [ ] @alexampa
> - [ ] @alexanderhmw
> - [ ] @alexandrx
> - [ ] @amc-nu
> - [ ] @andoh104
> - [ ] @anhnv-3991
> - [ ] @anhnv3991
> - [ ] @aohsato
> - [ ] @asimay
> - [ ] @aztk
> - [ ] @christopherho-ApexAI
> - [ ] @cirpue49
> - [ ] @ckirksey3
> - [ ] @colinsung
> - [ ] @daifuku07
> - [ ] @dejanpan
> - [ ] @droter
> - [ ] @drwnz
> - [ ] @eratostennis
> - [ ] @esteve
> - [ ] @francisvac
> - [ ] @h-ohta
> - [ ] @hakuturu583
> - [ ] @hatem-darweesh
> - [ ] @jacoblambert
> - [ ] @jitrc
> - [ ] @kfunaoka
> - [ ] @kitsukawa
> - [ ] @kuriking
> - [ ] @manato
> - [ ] @masa1228masa
> - [ ] @messi49
> - [ ] @mktk1117
> - [ ] @n-patiphon
> - [ ] @niwasaki
> - [ ] @ompugao
> - [ ] @pdsljp
> - [ ] @s-azumi
> - [ ] @sgermanserrano
> - [ ] @sujiwo
> - [ ] @syohex
> - [ ] @syouji
> - [ ] @travelbureau
> - [ ] @tushar-dadlani
> - [ ] @yk-fujii
> - [ ] @yn-mrse
> - [ ] @yukitsuji
> - [ ] @yukkysaito

+1

esteve, 2018-12-03:
@anhnv-3991 @anhnv3991 are these two accounts the same? Could reply with +1 from @anhnv-3991 to confirm they are? Thanks.

esteve, 2018-12-03:
@KenjiroYamada @MasatoTange @TomohitoAndo @alexanderhmw @alexandrx @colinsung @daifuku07 @droter @h-ohta @hatem-darweesh @hiro-ya-iv @jacoblambert @jitrc @kuriking @manato @masa1228masa @messi49 @mktk1117 @pdsljp @shinpei0208 @sujiwo @syohex @syouji @travelbureau @tushar-dadlani @yk-fujii @yn-mrse @yukkysaito unfortunately until you confirm that you're ok with the license change, we can can't switch to Apache. Could you posting a comment with "+1" if you agree with the change? Thanks.

anhnv3991, 2018-12-03:
@esteve They are the same. Please remove that anhnv-3991. I only use anhnv3991 for the Autoware project.

kfunaoka, 2018-12-04:
@esteve I've listed the people who work(ed) under the contracts under TierIV and/or Nagoya University. I think TierIV and Nagoya University can assume that they agree the license change. Is it insufficient to assume the agreement?

@KenjiroYamada @MasatoTange @TomohitoAndo @daifuku07 @h-ohta @hatem-darweesh @hiro-ya-iv @jacoblambert @kuriking @manato @masa1228masa @messi49 @pdsljp @shinpei0208 @sujiwo @syohex @syouji @yk-fujii @yukkysaito
Would you mind telling us if any comments?

esteve, 2018-12-04:
@kfunaoka thanks. I'd say that if they were employed by Tier IV and/or Nagoya University and their contract stated that the agreed to transfer their copyright to their employer, we should be fine. In any case, could you add a comment with their usernames? It's just so we can link each of the checkboxes to a comment, otherwise anyone can tick off whatever checkboxes they want.

kfunaoka, 2018-12-14:
> In any case, could you add a comment with their usernames?

@esteve The users listed in https://github.com/CPFL/Autoware/issues/1707#issuecomment-443987814. I assume each contract do not state that explicitly.

@shinpei0208 I think some people especially who worked on Autoware before might not see this issue. Should this issue be reviewed by legal process in Autoware Foundation?

kfunaoka, 2019-01-07:
Let's transit to Apache License 2.0!

@ChenxiTU @KenYN @KenjiroYamada @Mani-Sadhasivam @MasatoTange @TomohitoAndo @Tomoya-Sato @YamatoAndo @alexampa @alexanderhmw @alexandrx @amc-nu @andoh104 @anhnv-3991 @anhnv3991 @aohsato @asimay @aztk @christopherho-ApexAI @cirpue49 @ckirksey3 @colinsung @daifuku07 @dejanpan @droter @drwnz @eratostennis @esteve @francisvac @h-ohta @hakuturu583 @hatem-darweesh @hiro-ya-iv @jacoblambert @jitrc @kfunaoka @kitsukawa @kuriking @manato @masa1228masa @messi49

kfunaoka, 2019-01-07:
Let's transit to Apache License 2.0!

@mktk1117 @n-patiphon @niwasaki @ompugao @pdsljp @s-azumi @sgermanserrano @shinpei0208 @sujiwo @syohex @syouji @travelbureau @tushar-dadlani @yk-fujii @yn-mrse @yukitsuji @yukkysaito

esteve, 2019-01-07:
Fixed in #1741

closed 2019-01-07 by esteve