autowarefoundation/autoware #1723: [Bug] Fusion - Tracking pipeline not well connected

cirpue49, 2018-11-16:
# Bug
## Expected Behavior
Tracking /detection/combined_objects coming from range_vision_fusion nodes. 

## Actual Behavior
Cannot track properly.
[image]

## Steps to Reproduce the Problem
[image]

  * Launch vision_darknet_yolo3, ray_ground_filter, lidar_euclidean_clustrer_detect, calibration publisher, range_vision_fusion, imm_ukf_pda_track. 
 * Confirmed in Moiyama data. 

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: master

## This issue could be breaken into several issues.
* range_vision_fusion
  * Sometimes publish `camera` frame_id  instead of `velodyne` frame_id in DetectedObjects
  [image]

  * Sometimes publish empty pose DetectedObject 
  [image]

* imm_ukf_pda_track
  * Something messed up inside the tracker.

amc-nu, 2018-11-16:
Seems you’re not using topic synchronization.

kargarisaac, 2018-11-26:
@cirpue49 Which rosbag file do you use? The demo rosbag of autoware doesn't contain images.

cirpue49, 2019-01-31:
Still exists.
I enabled topic synchronization.
Got object filled with 0.
[image]

Can @amc address this issue when you have time?
I think you are not intended to publish 0-filled object from the fusion node.

jean-lucas, 2019-02-28:
Any updates on this? I have similar detection pipeline, and pose.position is always set to 0.

closed 2019-03-13 by k0suke-murakami