autowarefoundation/autoware #1725: Fail to  start grid_map_filter_visualization

Tomoya-Sato, 2018-11-21:
# Bug
## Expected Behavior
```rosdep update && rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO```
should install ```ros-kinetic-grid-map-visualization```

## Actual Behavior
It does not install ```ros-kinetic-grid-map-visualization```, preventing grid_map_filter_visualization from starting.
Manually running ```sudo apt-get install ros-kinetic-grid-map-visualization ``` will resolve the issue
## Steps to Reproduce the Problem

  1.Run according to [wiki](https://github.com/CPFL/Autoware/wiki/Source-Build)

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic
  - Autoware branch: feature/improve_astar_planner

hakuturu583, 2018-12-06:
@Tomoya-Sato rosdep read description from package.xml.
Have you add lines below to  package.xml??

```
 <build_depend>grid_map_visualization</build_depen>
 <build_export_depend>grid_map_visualization</build_export_depen>
 <exec_depend>grid_map_visualization</exec_depen>
```

Tomoya-Sato, 2018-12-14:
@hakuturu583 No. I just did `catkin_make`, right after checkout this branch.

aohsato, 2018-12-14:
@Tomoya-Sato Sorry for late, I fixed like above @hakuturu583 's suggestion. #1780
@hakuturu583 Actually so, I asked him to create this issue as my reminder. Thanks for your comment. :)

Plz review?

PR #1780 (2018-12-17): [Fix] Add grid_map_vizualization to run_depend
## Status
**DEVELOPMENT**

## Description
Fix #1725 

## Steps to Test or Reproduce
```
cd Autoware/ros
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
roslaunch object_map grid_map_filter.launch
```
closed 2018-12-18 by Tomoya-Sato