autowarefoundation/autoware #1728: Selecting image topic in calibration_publisher node

mitsudome-r, 2018-11-22:
After the update on #1595, we can select a topic name through combo selection box on RTM.
This misleads the selection of topic name for calibration_publisher node.

Calibration_publisher takes in "camera_id" and "image_topic" and subscribes to the combined topic name, but the new feature allows to select image topic with camera_id already attached. 
Therefore, setting camera_frame and choosing the correct image topic makes the node to read 
wrong image topic (e.g. camera_id/camera_id/topic_name)

I also would like to know why we have to subscribe to an image to publish camera_info in the first place.
Can't we publish at the start and latch it without reading an image?

amc-nu, 2018-11-27:
@mitsudome-r 
*Camera ID* stands for the *namespace*
*image topic source* was added since originally the image topic source was hard-coded to `image_raw`. However, this topic is not always available. (`image_color`, `image_mono`, etc)

> Can't we publish at the start and latch it without reading an image?
Latching caused several problems with synchronization. That's why the behavior was changed.

mitsudome-r, 2018-11-27:
@amc-nu 
Thank you for your reply.

> Camera ID stands for the namespace
image topic source was added since originally the image topic source was hard-coded to image_raw. 

I understand. Then one of the solution might be modifying runtime manager to not to use combo box for this particular topic. The combo box only allows you to select the topic name with namespace already attached, and it is kind of misleading.

> Latching caused several problems with synchronization. That's why the behavior was changed.

Could you explain a bit more about this? Do you mean you had some cases where you wanted to synchronize CameraInfo topic with other topics? I thought CameraInfo values are static and doesn't really need any synchronization.

amc-nu, 2018-11-27:
Yes I agree. RTM needs to be changed. That combo box allows to launch a publisher for each namespace..

> I thought CameraInfo values are static and doesn't really need any synchronization.

You’re right values should not change., However, time stamp should be updated. Calibration methods, and fusion methods require image and intrinsic topics to be synced. 
For instance, fusion only reads intrinsics(like you mentioned ), but also tf and detection results. If the intrinsics are latched then there is no trigger from the image side and, the time stamp cannot be matched with the corresponding tf and the detection results from image and lidar.
This is a simple and real example.

mitsudome-r, 2018-11-27:
> the time stamp cannot be matched with the corresponding tf and the detection results from image and lidar.

If tf is the only problem, using static tf broadcaster might solve the problem. http://wiki.ros.org/tf2/Tutorials/Writing%20a%20tf2%20static%20broadcaster%20%28C%2B%2B%29

So the best solution might be:
1.  Modify calibration_publisher to latch published CameraInfo message.
2.  Modify calibration_publisher to publish static tf instead of normal tf.
3.  Modify all the nodes that use timestamps from CameraInfo topic for synchronization. (Use timestamp from image topic instead)

However, Step 3. might require some work to list up all the nodes, so I guess the easiest work around for now is just fixing the RTM. 
I would like to fix this in the future though, since it is inefficient to publish same message & tf at high frequency.

closed 2020-05-29 by gbiggs