autowarefoundation/autoware #1731: Runtime manager can't run top utility for some locales

loafer-mka, 2018-11-22:
# Bug
The runtime manager fails to run top utility when locale settings assume comma as fractional separator in floating values; so CPU and memory usage cannot be collected and remains blank.

## Expected Behavior
actual information about CPU load and memory usage

## Actual Behavior
blank cpu indicators and memory usage; see image.

[image]

## Steps to Reproduce the Problem

  1. install autoware
  2. export LC_ALL=ru_RU.utf8
      (for example)
  3. cd Autoware/ros; ./run

## Specifications
   many configurations

## Possible fix
```
From 47f84d2f52a17409daab5a135d9cf04e9e53bdb6 Mon Sep 17 00:00:00 2001
From: Andrey Makarov <andrey.makarov@nami.ru>
Date: Thu, 22 Nov 2018 12:17:13 +0300
Subject: [PATCH] Fix locale settings before run top utility

---
 ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py b/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py
index 48f8f20..312416e 100755
--- a/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py
+++ b/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py
@@ -1418,7 +1418,7 @@ class MyFrame(rtmgr.MyFrame):
 		cpu_n = get_cpu_count()
 
 		while not ev.wait(interval):
-			s = subprocess.check_output(['sh', '-c', 'env COLUMNS=512 top -b -n 2 -d 0.1']).strip()
+			s = subprocess.check_output(['sh', '-c', 'env COLUMNS=512 LC_ALL=C top -b -n 2 -d 0.1']).strip()
 			i = s.rfind('\ntop -') + 1
 			s = s[i:]
 			wx.CallAfter(self.label_top_cmd.SetLabel, s)
-- 
2.7.4
```

closed 2020-05-29 by gbiggs

thiagodeborba, 2020-11-18:
How can I solve this problem?

loafer-mka, 2020-11-18:
I did add short patch there;
 
Really ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py
gets statistic using top utility:
s = subprocess.check_output(['sh', '-c', 'env COLUMNS=512 top -b -n 2 -d 0.1']).strip()
This utility prints numbers using current locale, so you may change locale before run utility:
 
s = subprocess.check_output(['sh', '-c', 'env COLUMNS=512 LC_ALL=C top -b -n 2 -d 0.1']).strip()
 
short patch was posted at https://github.com/Autoware-AI/autoware.ai/issues/1731

> 
>How can I solve this problem?
 
 
-----------
Regards,
Andrey