autowarefoundation/autoware #1746: [Discussion] Data Collection of Ground Truth in Simulator.

hakuturu583, 2018-11-28:
Now, we discuss simulation interface with Autoware in this issue. #1724 
In this issue, I recognize logging / publishing ground truth data is very very important in Simulator.
So, I want to discuss and define the design of the logging / publishing ground truth data.

hakuturu583, 2018-11-28:
I think it is requirements for collecting ground truth.

1. All logged data should be published on world or map frame.(static frame)
1. Publish ground pose tf data. (I will develop the ROS node which reads /simulation/ground_truth/($prefix)/vehicle_pose topic and publish ground truth frames.)

mitsudome-r, 2018-11-29:
@hakuturu583 
I have few questions regarding your uploaded image.
1. Why do you include $(prefix)/$(sensor_name) namespace for object_2d, whereas object_3d?

2. What kind of information is included in object_3d? Does it include all the objects in the world including buildings, signs, etc.? I think including static objects is over-informed and moving objects are enough.

hakuturu583, 2018-11-29:
@mitsudome-r Thanks for your pointing out.

>Why do you include $(prefix)/$(sensor_name) namespace for object_2d, whereas object_3d?

object_2d was object ROIs in camera, so  $(prefix)/$(sensor_name) namespace is necessary.
However, object_3d data was logged in world or map frame and include all vehicles and pedestrians in the simulator.
So,  $(prefix)/$(sensor_name) namespace is unnecessary.

>What kind of information is included in object_3d? Does it include all the objects in the world including buildings, signs, etc.? I think including static objects is over-informed and moving objects are enough.

DetectedObjectsArray includes object twist, pose, etc...

closed 2020-05-29 by gbiggs