autowarefoundation/autoware #1758: kitti_player launch bug or config issue '//'

LukeAI, 2018-12-03:
# Bug
## Expected Behavior
kitti_player launches and runs without errors

## Actual Behavior
kitti_player launches, begins seting up topics, then throws an error and dies:
IOError: [Errno 2] No such file or directory: '/home/luke/Autoware1.9/Autoware/ros/src/util/packages/kitti_pkg/kitti_player/dataset/2011_09_26/2011_09_26_drive_0093_sync//tracklet_labels.xml'
and dies

- tracklet_labels.xml does exist under this path without the miscellanaous double '//' at the end - so presumably there is a typo in a config file somewhere?

## Steps to Reproduce the Problem

  1. clone and build Autoware 1.8 or 1.9
  2. download kitti data and create a sym link to it as described in the instructions: https://github.com/CPFL/Autoware/tree/master/ros/src/util/packages/kitti_pkg
  3. source the Autoware environment
4. Run roslaunch kitti_launch kittiplayer.launch fps:=10

## Specifications

  - Ubuntu version: 16.04
  - ROS version: 1.12.14
  - Autoware branch: same result in 1.8, 1.9

closed 2018-12-07 by LukeAI

InitialDuan, 2021-01-26:
Hi @LukeAI, how did you solve this issue? I'm facing the same problem.

LukeAI, 2021-01-26:
from my notes...
```
# create a symbolic link to the directory containing the raw kitti data from the package eg. something like
ln -s ~/projects/kitti_raw ~/aiPE/ros/src/util/packages/kitti_pkg/kitti_player/dataset

# Modify the `src/util/packages/kitti_pkg/kitti_launch/launch/kittiplayer.launch` file to choose the set you wish to reproduce ("date" and "kitti_data")

#cd to the workspace root and build the required packages in this order:
catkin_make --pkg kitti_player
catkin_make --pkg points2image
catkin_make --pkg calibration_camera_lidar
```

JWhitleyWork, 2021-01-26:
Autoware.AI uses `colcon` rather than `catkin_make` so you can use `colcon build --packages-up-to calibration_camera_lidar` instead.

LukeAI, 2021-01-26:
right, my notes only apply to a really old autoware, idk if they will help.

InitialDuan, 2021-01-26:
Hi @JWhitleyWork, that does not work for me. I provide more infomation on [ROS answer](https://answers.ros.org/question/370350/autoware-unable-to-launch-kitti_player-properly/), could you have a look?