autowarefoundation/autoware #1763: Is autoware running without IMU?

randomblbl, 2018-12-04:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior

## Actual Behavior

## Steps to Reproduce the Problem

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS version:
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

hakuturu583, 2018-12-05:
Autoware can be used without IMU.
But, lidar sensor is necessary.

hakuturu583, 2018-12-06:
Now, we are not using yaw rate for low level control.

randomblbl, 2018-12-06:
> Now, we are not using yaw rate for low level control.

Actually, I can not find the lower level controller in Autoware code.
The Autoware produces twist_cmd with velocity and angular velocity, but I could not find the drive-by-wire module (lower-level controller) which controls the brake pedal, accelerate pedal and steering to meet the required speed and angular velocity.

Do you know where I could find it in Autoware  ?

kfunaoka, 2018-12-08:
@randomblbl There is no drive-by-wire module in Autoware. Autoware uses drive-by-wire modules that each car provides.

closed 2018-12-10 by randomblbl