autowarefoundation/autoware #1764: Any plan to use multi-sensor fusion in localization

junjieshen, 2018-12-05:
Hi,

Is there any plan to incorporate multi-sensor fusion in localization? 

Currently, Autoware seems only use NDT matching for localization. Although the codebase does support GPS and IMU, it seems GPS is only used during initialization, and IMU is disabled by default.

AFAIK, multi-sensor fusion (IMU+GNSS+Lidar) is a common approach in autonomous vehicle localization. Other platforms like Apollo used a Kalman Filter to fuse GNSS and Lidar together with IMU. And even Udacity includes that as a major part of the self-driving car course program. So I was wondering if there is any plan to use some fusion algorithms, such as Kalman filter or particle filter, in the future? If not, what would be the difficulty/consideration prevent you from doing that?

Thanks a lot!
Junjie

hakuturu583, 2018-12-05:
NDT matchig can use imu and GPS data.
Open runtime manger and configure options.

junjieshen, 2018-12-05:
Thanks for the prompt reply!

For GPS, do you mean the [gnss callback](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/lidar_localizer/nodes/ndt_matching/ndt_matching.cpp#L535)? It looks like GNSS is only invoked during initialization, or when NDT matching fitness is >500 (though I don't know how often the fitness will go over this threshold).

And for IMU, I do see the [IMU callback](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/packages/lidar_localizer/nodes/ndt_matching/ndt_matching.cpp#L848), but I couldn't find where the IMU results are used. Could you please point me to any relevant code which is applying IMU results to the localizer_pose? 

Thanks!

hakuturu583, 2018-12-06:
Autoware set IMU data to static parameters "imu" in line 236.
And predict pose from "imu" parameter.

amc-nu, 2019-05-31:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-05-31 by amc-nu