autowarefoundation/autoware #1765: timestamp of velodyne packets does not changed

maedaseiya, 2018-12-05:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
```
rostopic echo /velodyne_packets/header
seq: 0
stamp: 
  secs: 1543996478
  nsecs: 959228516      <--------nsec 
frame_id: velodyne
---
seq: 1
stamp: 
  secs: 1543996479
  nsecs:  60075283   <----------nsec  is changed,usually.
frame_id: velodyne
---
```

## Actual Behavior
```
rostopic echo /velodyne_packets/header
seq: 4699
stamp: 
  secs: 904636608      
  nsecs:     32663       <--------nsec 
frame_id: "velodyne"
---
seq: 4700
stamp: 
  secs: 904636608    <----------sec  is not changed and  it should be about 1543996479.
  nsecs:     32663      <----------nsec is not changed .
frame_id: "velodyne"
---
```
## Steps to Reproduce the Problem

  1.connect VelodyneVLP-16
  2. start Runtime manager and select sencing tab and  click  Velodyne VLP-16
  3. check  time stamp of /velodyne_packets.

it also happened Velodyne32C
It always occurs on a specific PC.

## Specifications

  - Ubuntu version:16.04
  - ROS version:kinetic
  - Autoware branch:master 1.9.1 Autoware branch:master 1.6.1

kfunaoka, 2018-12-06:
@maedaseiya told me that the following command fixed the problem. I'm not sure why.

1. touch ~/Autoware_1.9.1/ros/src/sensing/drivers/lidar/packages/velodyne/velodyne_driver/src/lib/input.cc
1. cd ~/Autoware_1.9.1/ros
1. source devel/setup.sh
1. catkin_make

amc-nu, 2019-01-04:
Make sure you don't have the lidar configured to double returns.

closed 2019-01-04 by amc-nu