autowarefoundation/autoware #1766: how to use ndt_matching for locate using my own bag file?

defendergz, 2018-12-06:
hi everyone,I want to use ndt_matching for locate using my own bag file, just like this:https://www.youtube.com/watch?v=sul-osvg42A , i have got correct result like the video by using sample_moriyama_150324.bag and data directory in demo.But when I modify some parameter and launch file for my own bag file for locate, it doesn't work like video above. there are nothing on rviz.

# Bug
## Expected Behavior
i modify same parameter and launch in my own bag file, use ndt_matching and i wanna get output ndt_matching.csv ,which contain pose.x,y,z roll,pitch,yaw etc. and look something such as point_map and scan_point(in demo,it is /point_raw).

## Actual Behavior
 i can't see anything on rviz, and there isn't any ndt_matching.csv output.

## Steps to Reproduce the Problem
this is my .bag's content:
my_bag20181101.bag:
type: 
nav_msgs/Odometry
sensor_msgs/Imu
sensor_msgs/NavSatFix
sensor_msgs/PointCloud2
topics:
/compensated_imu   : sensor_msgs/Imu
/fix:              sensor_msgs/NavSatFix
/lidar_points:        sensor_msgs/PointCloud2
/odom:        nav_msgs/Odometery
my actual steps:
1 ./run----play my_bag20181101.bag----pause
2 setup---Velodyne---TF(0 0 0 0 0 0)---vehicle model(blank)
3 Map---pointCloud(load my own pcd file)---TF(load my_tf.launch modified)---i have no Vector map so i skip this.
4 Sensing---voxel_grid_filter---app---set Point topic from /points_raw to /lidar_points.
5 Computing--- √ nmea2tfpose--- √ ndt_matching---app--- select GNSS---select Predict Pose OFF---select pcl_generic---set imu_topic to /compensated_imu
[image]

6 Rviz---file---open config---ros/src/.config/rviz/default.rviz---add PointCloud2 select topic:/lidar_points (others is default).
[image]

7 click pause
NOTE: in step 3,my_tf.launch is:( i modify parameters world_to_map for my localation)
[image]

my screen:(i got a status error: For frame [rslidar]: Frame [rslidar] does not exist)
[image]

## Specifications

  - Ubuntu version:16.04 
  - ROS version:kinetic
  - Autoware branch: master

anyone can help me ?

hakuturu583, 2018-12-06:
@defendergz 
Please check below.

1. Did you play rosbag first??
When you play rosbag first, Autoware set use_sim_time parameter true.
1. Did you enable base_link_to_localizer node??
1. rs lidar node is exist??

mitsudome-r, 2018-12-06:
It looks like you don't even see /points_map, so showing pointcloud map on rviz properly might be a good step.
This problem could be caused by few reasons:
1. Your tf.launch is wrong. Make sure that your loaded pcd file has points that are somewhat near (457099, 4404975).
2. Your rviz config is wrong. Make sure a) rviz "Fixed Frame" under Global Options is set to world and b) Target Frame in Views panel is set to <Fixed Frame>. Then, press Zero button in Views panel and scroll back to zoom out. If you still don't see your map pointcloud, then try publishing pcd again.

Also, TF button under Setup Tab only sets "base_link"->"velodyne" frame. I assume your lidar_points topic is published in rslidar frame so you need to publish "base_link"->"rslidar" tf on your own. 
e.g.
`rosrun tf static_transform_publisher 0 0 0 0 0 0 /base_link /rslidar 10
`

defendergz, 2018-12-06:
@hakuturu583 
> Please check below.
> 
> 1. Did you play rosbag first??
>    When you play rosbag first, Autoware set use_sim_time parameter true.
> 2. Did you enable base_link_to_localizer node??
> 3. rs lidar node is exist??

thanks a lot for your  prompt reply.

1.i do play .bag first and then quickly click pause as step 1 above.
2. i enable base_link_to_localizer in setup tab as step 2 above.but i modify parameter from (1.2 0 2)  to below:
[image]
3. i check rosnode using rosnode list during run my own bag:(rs lidar node don't exist),(13 nodes)
[image]
 BUT, then I run autoware demo sample_moriyama_150324.bag, i found rosnode is:(14 nodes)
[image]

there isn't anything like velodyne node exist. but the demo is work well and get correct result like video above. 
1.could you please tell me if i have to write a rslidar node or not?  
2.And i also found during run my own bag, there isn't /vector_map_loader exist, how should I solve this? or is it necessary?

defendergz, 2018-12-06:
@mitsudome-r 
thanks for your reply
> It looks like you don't even see /points_map, so showing pointcloud map on rviz properly might be a good step.

I add PointCloud2 /points_map, and Points Map /points_map by default.rviz:
[image]

> This problem could be caused by few reasons:
> 
> 1. Your tf.launch is wrong. Make sure that your loaded pcd file has points that are somewhat near (457099, 4404975).

I sure that pcd file loaded contains points near (457099,4404957)

> 2. Your rviz config is wrong. Make sure a) rviz "Fixed Frame" under Global Options is set to world and b) Target Frame in Views panel is set to . Then, press Zero button in Views panel and scroll back to zoom out. If you still don't see your map pointcloud, then try publishing pcd again.

i do as your advices:
[image]
[image]

> Also, TF button under Setup Tab only sets "base_link"->"velodyne" frame. I assume your lidar_points topic is published in rslidar frame so you need to publish "base_link"->"rslidar" tf on your own.
> e.g.
> `rosrun tf static_transform_publisher 0 0 0 0 0 0 /base_link /rslidar 10 `

[image]

 i don't press TF under Baselink to Localizer
and then i do this:
[image]

then i get a error :  world  fail to connect to rslidar 
[image]
and also there is nothing on Rviz

mitsudome-r, 2018-12-06:
try changing target frame in current view from "base_link" to "Fixed Frame" and press zero. Do you see pointcloud map?.
Also, can you paste the result of `rosrun rqt_tf_tree rqt_tf_tree`?( make sure you press refresh button)

mitsudome-r, 2018-12-06:
> And i also found during run my own bag, there isn't /vector_map_loader exist, how should I solve this? or is it necessary?
You don't need /vector_map_loader for localization so you don't have to worry about it.

mitsudome-r, 2018-12-06:
sorry I closed the issue by accident

ajay1606, 2018-12-07:
@mitsudome-r Are you sure, for localization vector map not necessary! Can we localize the vehicle just using point cloud map and real-time LiDAR data?

Could you please comment on this? I really appreciate your feedback.
Then, where does the vector map useful? NDT_Mathcing doesn't need vector map?

defendergz, 2018-12-07:
@mitsudome-r 
when i perform:
`rosrun tf static_transform_publisher 0 0 0 0 0 0 /base_link /rslidar 10`
it seems work,and this is rqt_tf_tree:
[image]
then i run rviz:(as same steps as above),i get this:
[image]
i think this connection error is doesn't matter,because it also get the same error when i run demo like videos.what's your opinion?or do u think so?and then in order to resolve it,i modify base_link :(because it is world to map,and map to mobility in tf.launch of demo)
`rosrun tf static_transform_publisher 0 0 0 0 0 0 /mobility /rslidar 10`
although I still think it's dispensable.and this is rviz on my screen:
[image]

it does work ! thanks a lot.
but mobility/rslidar/base_link is underground.in other words,they all under map's ground points. 
anyone can solve this? is it cause by map that i loaded pcd file has no points near by  (457099,4404957)?
Anyway, thank you again for your help. @mitsudome-r

mitsudome-r, 2018-12-07:
@ajay1606 
Only pointcloud map and real-time lidar data are necessary for localization. Vector map is needed for other features such as stopping at stopline, detecting signal lights, switching blinkers at intersection, and so on.

mitsudome-r, 2018-12-07:
@defendergz 
> then in order to resolve it, i modify base_link :(because it is world to map,and map to mobility in tf.launch of demo) rosrun tf static_transform_publisher 0 0 0 0 0 0 /mobility /rslidar 10

Please ignore mobility frame since it is not used for localization.
It needs to be base_link to rslidar frame in order for the localization to work.

According to your tf_tree, ndt_matching is not working. (It should publish tf from map to base_link)
Is filtered_points published? 
If you pointcloud map is large o dense, ndt_matching takes some time to initialize. See system monitor and wait until cpu usage goes down after you start ndt_matching. Then, try giving initial pose from 2D Pose Estimate on RVIZ.

defendergz, 2018-12-07:
@mitsudome-r 
if i ignore mobility frame,i will get error following:
[image]

because I loaded tf.launch under Map Tab is /world to /map and /map to /mobility,and then rosrun tf static_transform_publisher 0 0 0 0 0 0 /base_link /rslidar 10,so there isn't a connection between world and rslidar. Should I still ignore this error?

> According to your tf_tree, ndt_matching is not working. (It should publish tf from map to base_link)
Is filtered_points published?

there are two branches in tf tree,and there isn't connection exist. 
>  (It should publish tf from map to base_link)

do you mean i need to add a command again like: `rosrun tf static_transform_publisher x x x x x x /map /base_link 10`   ?

mitsudome-r, 2018-12-07:
>there are two branches in tf tree,and there isn't connection exist.
do you mean i need to add a command again like: rosrun tf static_transform_publisher x x x x x x /map /base_link 10 ?

No, you don't need additional nodes. The branches are unconnected because ndt_matching is not working properly. You have to find out the reason for that.

Have you checked the following?
>Is filtered_points published?
If you pointcloud map is large or dense, ndt_matching takes some time to initialize. See system monitor and wait until cpu usage goes down after you start ndt_matching. Then, try giving initial pose from 2D Pose Estimate on RVIZ.

defendergz, 2018-12-07:
@mitsudome-r  Thank you again for your reply.
I've figured out the role of ndt_matching in tf tree. 
> Is filtered_points published?
If you pointcloud map is large or dense, ndt_matching takes some time to initialize. See system monitor and wait until cpu usage goes down after you start ndt_matching. Then, try giving initial pose from 2D Pose Estimate on RVIZ.

filtered_points has published,use echo filtered_points i get many value of point.
my pcd file is only 50+MB.
1.there are sensor_msgs/NavSatFix contained in my bag file , and i select fix2tfpose under Computing Tab,as a result, i don't press 2D Pose Estimate to initial pose. Is this feasible? if yes,  how to select plane    number in app?
2. i watch cpu usage util it goes down(about 55%),and there are nothing on RVIZ,and get the same error above:not find a connection. and i waited for about 1 minute,error remain the same.but when run demo provided by autoware,it work well.
3. also i try not to select fix2tfpose, and initial pose by 2D pose Estimate, still get the same error.

mitsudome-r, 2018-12-08:
>1.there are sensor_msgs/NavSatFix contained in my bag file , and i select fix2tfpose under Computing Tab,as a result, i don't press 2D Pose Estimate to initial pose. Is this feasible? if yes, how to select plane number in app?

Where did you take this bag file? Current fix2tfpose only supports gps signals recorded in Japan. Otherwise, you need to give initial pose through RVIZ 2D Pose Estimate.

Also, for odometry, ndt_matching will look for /vehicle/odom so you probably have to do 
`rosrun topic_tools relay /odom /vehicle/odom`

defendergz, 2018-12-09:
@mitsudome-r thank u reply
i use this bag in china. so do i have to initial by 2D Pose Estimate? and in ndt_matching,does its accuracy depend on the location of initialization? In other words, is it related to the location of my mouse click?

mitsudome-r, 2018-12-10:
>i use this bag in china. so do i have to initial by 2D Pose Estimate? 
Yes. 

> in ndt_matching,does its accuracy depend on the location of initialization? In other words, is it related to the location of my mouse click?
No, once ndt_matching fits properly, the accuracy of localization does NOT depend on the location of initialization.

defendergz, 2018-12-10:
@mitsudome-r thank  u so much for  your series of answers.

closed 2018-12-10 by defendergz

ajay1606, 2018-12-13:
@defendergz @mitsudome-r Thank a lot for your discussion, it really helped me to understand a lot of things from here.  Although I followed your comments, I stuck with a similar issue. Could you please help me solving this issue. Also, I have posted my issue in #1773 

Version:
OS: Ubuntu 16.04
ROS: Kinetic
Autoware: 1.9.0
LiDAR:
velodyne 64 channel
[image]
[image]
[image]
[image]

I really appreciate your sincere reply. Thanks in advance.

Glidepathuav, 2018-12-13:
Sidebar

Anyone interested in a Velodyne vlp16 lidar. Brand new?

ajay1606, 2018-12-14:
@defendergz @mitsudome-r , Now i am able to visualize the points_raw in the rviz, but the issues is that, it's not mapping correctly on the map. Could you please share your suggestions here. I am looking forward to hear from you. 
[image]

[image]

HongweiSunny, 2019-03-10:
@defendergz  Hi, have you solved you problem? I have the same question. And I got my bag file in China too. Looking forward to your reply.