autowarefoundation/autoware #1772: [Feature] costmap_generator

cirpue49, 2018-12-11:
# New Feature: costmap_generator
### Please refer to this issue about high-level diagram. https://github.com/CPFL/Autoware/issues/1719#issuecomment-438632085

The costmap_generator node will 
subscribe: 
/points_no_ground (sensor_msgs::PointCloud2)
/prediction/moving_predictor/objects (autoware_msgs::DetectedObjectArray)
/tf
vector_map

publish:
/semantics/costmap (grid_map::GridMap)
/semantics/costmap_generator/occupancy_grid(nav_msgs::OccupancyGrid)

Expected behavior: Generate useful costmap for the planning module. For example, https://github.com/CPFL/Autoware/pull/1594

kennedywai, 2018-12-11:
Nice, is this available now to for testing the node?

kennedywai, 2018-12-11:
I have been planning not to use vector map to achieve Hybrid A* and trying to build a global costmap like ROS used to do in ROS Navigation Stack.

kennedywai, 2018-12-13:
@cirpue49  Thank you for posting this! I will find a time to test this later on! BTW is this costmap generator only for local costmap instead of converting point cloud map to global costmap?

cirpue49, 2018-12-18:
What do you mean by "only for local costmap instead of converting point cloud map to global costmap" ?

cirpue49, 2019-01-07:
@kennedywai 
Sorry for the delay. 
If you mean "static costmaps" by generating costmap for static objects, it can be possible.

You can subscribe to the static objects. Topic: "/detection/lidar_detector/objects" from euclidean cluster.

closed 2019-03-12 by k0suke-murakami