autowarefoundation/autoware #1773: ndt_matching : unable to map bag file Points_Raw to world frame

ajay1606, 2018-12-12:
Hello everyone,
I am trying to use ndt_matching for locating a vehicle without GNSS using own rosbag file just like shown in the video https://www.youtube.com/watch?v=ODlxMzGTJzw. Its working good with demo datasets sample_moriyama_150324.bag provided by the Autoware. But its not working with our dataset. Even i am unable to visualize the Points_Raw from the .bag file. 

Currently, Point cloud in the bag file recorded using the Velodyne 64 channel liDAR.  

# Bug
## Expected Behavior
I am looking to estimate the position of the vehicle (x, y,z, roll, pitch, yaw)

## Actual Behavior
Currently, I am able to visualize the Pointcloud map, vector map when i run ../my_map.launch &my_sensing.launch 

But when i run my_localization. launch, everything gets disappears from the rviz except vehicle model.

## Steps to Reproduce the Problem

  1. Play rosbag and pause
  2. Load Point cloud map, vector map and Tf.launch (Modified) (args="0 0 0 0 0 0 /world /map 10")
  3. Run my_map.launch, my_sensing.launch and my_localization.launch

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: Master Branch (1.9.0)

With Launching localization node:
[image]
Without Launching localization node:
[image]

Could anyone please help me solve this.

hakuturu583, 2018-12-17:
@ajay1606 It seems that the pointcloud map succeed to load.
But ndt matching does not works.
Please check below.
1. ndt matching node is working? (segmentation fault, error, etc..)
Please open another terminal and put this command.
```
rosnode list | grep ndt_matching
```
2. Are you set initial value?
ndt matching can use 2d pose estimate result or GNSS result as a initial value of ndt matching.
Are you set use_gpu option in runtime mangaer? or set initial pose using 2d pose estimate.

ajay1606, 2018-12-18:
@kitsukawa @hakuturu583, Thank you so much for your response..,
Yes, I have initialized the position as shown in the below, and able to check the points_raw on the point cloud map.
**Question 1:**
But, the localization seems to not working (ndt_matching) correctly, even I am not getting any error message. vehicle position going out of the expected path. Could you please comment on this. It will be a great help for us.

Here with attaching some results:
Position Initialization:
[image]
Initial Position visualization:
[image]
While Localization in progress:
[image]
While Localization in progress:
[image]
NDT_matching node:
[image]

**Question 2:**
Unable to initialize using GNSS:
Autoware Dataset:
[image]
GPS positions and Point cloud are in the same coordinate frame.

Our dataset:
[image]
GPS position data & the Points in the point cloud are completely different coordinate range. Hence we are unable to initialize the using GNSS, Could you please suggest us how to transofrm the GPS positions and point clouds in the common coordinate frame.

We are looking for your sincere response, Thanks in advance.

Kind regards,
AJay

hakuturu583, 2018-12-18:
@ajay1606 From your posted imaged, I think you set initial pose from 2d pose estimate.
I think 2 possible situation.
1. You set wrong initial pose.
Try setting correct initial pose.
2. VLP-64 takes too much ground point.
From your image, almost of all point cloud data comes from the ground.
So, I think the ndt matching algorithm matches sensor pointcloud data with pointcloud from ground.
Can you use ray_ground_filter and remove ground data from pointcloud data and input the pointcloud data to the ndt matching.

Best,
Masaya

hakuturu583, 2018-12-18:
Question 2 : 
[image]
Please enable use_gnss option.

ajay1606, 2018-12-18:
@hakuturu583 Thanks for your detailed response. I will verify the initial position again, and get back to you.

ajay1606, 2018-12-18:
@hakuturu583 I have observed another issue: Since i am using velodyne 64 channel, I have to modify the "my_sensing.launch" file.
If i change my my_sensing.launch file to velodyne_64 configuration (with velodyne 64 (.yaml file)) as below, i am unable to see the points in rviz
**//Velodyne64**
<launch>
  
  <arg name="velodyne_calib" default="$(env HOME)/.autoware/quick_start/hdl64_Es2.yaml"/>
<arg name="camera_calib" default="$(env HOME)/.autoware/data/calibration/camera_lidar_3d/prius/ nic-150407.yml"/>
  
  <include file="$(env HOME)/.autoware/quick_start/64e_s2_points.launch">
    <arg name="calibration" value="$(arg velodyne_calib)"/>
  </include>
  
  
  
  <include file="$(find autoware_pointgrey_drivers)/scripts/grasshopper3.launch">
    <arg name="CalibrationFile" value="$(arg camera_calib)"/>
  </include>
</launch>

**//Velodyne32**
If i use this configuration, i get velodyne points projected correctly without filtering
<launch>
  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/32db.yaml"/>
  <arg name="camera_calib" default="$(env HOME)/.autoware/data/calibration/camera_lidar_3d/prius/ nic-150407.yml"/>
  
  <include file="$(find velodyne_pointcloud)/launch/velodyne_hdl32e.launch">
    <arg name="calibration" value="$(arg velodyne_calib)"/>
  </include>
  
  
  
  <include file="$(find autoware_pointgrey_drivers)/scripts/grasshopper3.launch">
    <arg name="CalibrationFile" value="$(arg camera_calib)"/>
  </include>
</launch>
**After ray_ground_filtering** 
After the ground filtering points looks like this, Is this because of wrong.yaml file ?
[image]

Could you please suggest me correct launch configurations for the velodyne64 channel?

Kind Regards,
Ajay

hakuturu583, 2018-12-18:
@ajay1606 Can you show me all lines of your my_sensing.launch??

ajay1606, 2018-12-18:
**This is my edited velodyne 64 channel launch file**
```
<launch>
  <!-- calibration file path -->
  <arg name="velodyne_calib" default="$(env HOME)/.autoware/quick_start/hdl64_Es2.yaml"/>
<arg name="camera_calib" default="$(env HOME)/.autoware/data/calibration/camera_lidar_3d/prius/ nic-150407.yml"/>
  <include file="$(env HOME)/.autoware/quick_start/64e_s2_points.launch">
    <arg name="calibration" value="$(arg velodyne_calib)"/>
  </include>
  <!-- Javad Delta 3 -->
  <!-- <node pkg="javad_navsat_driver" type="gnss.sh" name="javad_driver"/> -->
  <!-- PointGrey Grasshopper3 -->
  <include file="$(find autoware_pointgrey_drivers)/scripts/grasshopper3.launch">
    <arg name="CalibrationFile" value="$(arg camera_calib)"/>
  </include>
</launch>
```
**//Velodyne 32 (Autoware)**
Below launch file works good only to non-filtered points, if i apply  ray_ground_filter, points appears to be wrongly projected. (As shown in the previous comment picture)
```
<launch>
  <!-- calibration file path -->
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/32db.yaml"/>
  <arg name="camera_calib" default="$(env HOME)/.autoware/data/calibration/camera_lidar_3d/prius/ nic-150407.yml"/>
  <!-- HDL-32e -->
  <include file="$(find velodyne_pointcloud)/launch/velodyne_hdl32e.launch">
    <arg name="calibration" value="$(arg velodyne_calib)"/>
  </include>
  <!-- Javad Delta 3 -->
  <!-- <node pkg="javad_navsat_driver" type="gnss.sh" name="javad_driver"/> -->
  <!-- PointGrey Grasshopper3 -->
  <include file="$(find autoware_pointgrey_drivers)/scripts/grasshopper3.launch">
    <arg name="CalibrationFile" value="$(arg camera_calib)"/>
  </include>
</launch>
```
Thanks

ajay1606, 2018-12-20:
@hakuturu583 , That's all that I am using in my_sensing. launch file and Is it there any issue with the my_sensing. launch file (64 channel lidar) ? could you please confirm on this!

hakuturu583, 2018-12-21:
@ajay1606 ＯＫ， Ｉ ｗｉｌｌ ｃｏｎｆｉｒｍ．

ajay1606, 2019-01-03:
@hakuturu583 New year wishes to you, By the way, have you got any chance to review my issues? Related to ray_ground_filter for 64 channel velodyne LiDAR data and NDT matching.

ajay1606, 2019-01-09:
@hakuturu583, ANy suggestions please, Looking forward to hearing from you!

hakuturu583, 2019-01-10:
@ajay1606 Sorry for my late reply.
Is the poins_raw topic is correct?
I think it is the velodyne driver bug.

ajay1606, 2019-01-10:
@hakuturu583, thank you for coming back, I was eagerly looking for your help. I think it's not Velodyne driver bug. Because of the following reasons;
1. points_raw in my case (topic: velodyne_points) are rendering correctly without applying ray_ground_filtering preprocessor.
2. Also, If i changed my Velodyne LIDAR 64 channel driver in the my_sensing launch file,  i am unable to see the points in the rviz. (currently, i using autoware version my_sensing_launch file). I have attached the my_sensing launch file in the previous comment. And could you please confirm on the correct launch file for my case .

ajay1606, 2019-01-10:
@hakuturu583 I really appreciate, if you could test on my data (https://drive.google.com/open?id=1HsabYoNX5nlZbo266os-cKKSLlyqpsjl). and please can you confirm me once?

ajay1606, 2019-01-13:
@hakuturu583 Have you got any chance to check my bag file? looking forward to hearing from you.

ajay1606, 2019-01-14:
@hakuturu583 , If you get chance, i request you to review the video once. Actual problem of localization with my dataset can be understood in this video.

[video_localization.zip](https://github.com/CPFL/Autoware/files/2755579/video_localization.zip)

Hope this will helps to give suggestions.

Kind Regards,
AJay

ajay1606, 2019-01-20:
@kitsukawa Could you please help me in fixing the issues with ndt_matching?

ajay1606, 2019-01-24:
@kitsukawa & @hakuturu583 Finally ndt_matching working well with velodyne_64 channel data. There was issues with my_sensing.launch file (.yaml file & .launch files). Now everything working well. Thanks for your continuous feedback.

closed 2019-01-24 by ajay1606

Tinade, 2020-10-02:
@kitsukawa i got the same problem can you help me. i am using i have collected .pcap file using VLP32C and converted to a point cloud and .bag file but when i tried to localize the ros bag playing is not on the map. someone please help me. i can email my .pcd file and bag for someone interested to help and the tf i used is the one in autoware/src/.config

JWhitleyWork, 2020-10-08:
@Tinade Please do not revive zombie threads from years ago. Instead, create a new question on [ROS Answers](https://answers.ros.org/) with the `autoware` tag.