autowarefoundation/autoware #1779: Invalid roslaunch XML syntax

pfrcks, 2018-12-14:
Hi,

I have installed Autoware using nvidia-docker2 and kinetic. When I'm trying to follow through the tutorial, I am able to start the manager, load the rosbag and start rviz. However after loading  the map i get the following error

```
[roslaunch][INFO] 2018-12-14 08:35:44,814: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2018-12-14 08:35:44,816: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2018-12-14 08:35:44,816: roslaunch starting with args ['/opt/ros/kinetic/bin/roslaunch', '/home/autoware/Autoware/docs/quick_start/my_map.launch']
[roslaunch][INFO] 2018-12-14 08:35:44,816: roslaunch env is {'ROS_DISTRO': 'kinetic', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/autoware/Autoware/ros/devel/share/common-lisp', 'LOGNAME': 'root', 'USER': 'root', 'HOME': '/home/autoware', 'PATH': '/opt/ros/kinetic/share/euslisp/jskeus/eus//Linux64/bin:/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin', 'CMAKE_PREFIX_PATH': '/home/autoware/Autoware/ros/devel:/opt/ros/kinetic', 'LD_LIBRARY_PATH': '/opt/ros/kinetic/share/euslisp/jskeus/eus//Linux64/lib:/home/autoware/Autoware/ros/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu', 'LANG': 'en_US.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'ROS_LOG_FILENAME': '/home/autoware/.ros/log/26111ba2-ff7b-11e8-ad88-2cfda1e413ec/roslaunch-Colossus-2517.log', 'XAUTHORITY': '', 'SHLVL': '2', 'DISPLAY': ':1', 'WINDOWID': '60817443', 'ROS_ETC_DIR': '/opt/ros/kinetic/etc/ros', 'SUDO_USER': 'autoware', 'USERNAME': 'root', 'EUSDIR': '/opt/ros/kinetic/share/euslisp/jskeus/eus/', 'PYTHONPATH': '/home/autoware/Autoware/ros/devel/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages', 'SUDO_UID': '1000', 'VTE_VERSION': '4205', 'ROS_ROOT': '/opt/ros/kinetic/share/ros', 'G_FILENAME_ENCODING': 'UTF-8', 'ARCHDIR': 'Linux64', 'PKG_CONFIG_PATH': '/home/autoware/Autoware/ros/devel/lib/pkgconfig:/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig', 'ROS_PACKAGE_PATH': '/home/autoware/Autoware/ros/src:/opt/ros/kinetic/share', 'SUDO_COMMAND': '/bin/bash run', 'SUDO_GID': '1000', 'LC_MESSAGES': 'POSIX', 'HOSTNAME': 'Colossus', 'PWD': '/home/autoware/Autoware/ros', 'ROS_MASTER_URI': 'http://localhost:11311', 'ROS_VERSION': '1', 'MAIL': '/var/mail/root', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:'}
[roslaunch][INFO] 2018-12-14 08:35:44,816: starting in server mode
[roslaunch.parent][INFO] 2018-12-14 08:35:44,817: starting roslaunch parent run
[roslaunch][INFO] 2018-12-14 08:35:44,817: loading roscore config file /opt/ros/kinetic/etc/ros/roscore.xml
[roslaunch][INFO] 2018-12-14 08:35:45,071: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2018-12-14 08:35:45,071: loading config file /home/autoware/Autoware/docs/quick_start/my_map.launch
[roslaunch][ERROR] 2018-12-14 08:35:45,071: while processing /home/autoware/.autoware/data/tf/tf.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/.autoware/data/tf/tf.launch'
[roslaunch][ERROR] 2018-12-14 08:35:45,072: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2018-12-14 08:35:45,072: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 453, in load_config_default
    raise RLException(e)
RLException: while processing /home/autoware/.autoware/data/tf/tf.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/.autoware/data/tf/tf.launch'

[rospy.core][INFO] 2018-12-14 08:35:45,072: signal_shutdown [atexit]
```

I checked the mentioned location and as expected cannot find the file `tf.launch` which is mentioned in `my_map.launch`

pfrcks, 2018-12-14:
Figured this out. Even though the shared directory was located at `/home/autoware/shared_dir` and contained the actual data, the launch file expected it to be at `/home/autoware/.autoware`

I created a symlink and it started working.

closed 2018-12-14 by pfrcks

baroselfdriving, 2020-02-04:
Was the same solution for me.. move from shared_dir to .autoware using the terminal screen (be careful, not the terminal in your host. the terminal screen after roslaunch command.